motioneye-project/motioneye #1190: Upgrade to 0.40rc3

nuiler, 2019-03-18:
How can I upgrade to version 0.40rc3, currently i use 

motionEye Version 0.39.3
Motion Version 4.1
OS version Raspbian 9.4

ccrisan, 2019-03-18:
`pip2 install --upgrade motioneye==0.40rc3` should do the job.

00lex, 2019-03-18:
or 
`pip install --pre motioneye --upgrade`

nuiler, 2019-03-19:
I have Successfully installed MarkupSafe-1.1.1 backports-abc-0.5 futures-3.2.0 jinja2-2.10 motioneye-0.40rc4 pillow-5.4.1 pycurl-7.43.0.2 singledispatch-3.4.0.3 six-1.12.0 tornado-5.1.1 

but in the webinterface displayed 

motionEye Version | 0.39.3
Motion Version | 4.1
OS Version | Raspbian 9.4

ccrisan, 2019-03-19:
@nuiler restart the server.

vplustute, 2019-03-26:
can I start the update by SSH?
I connected via SSH und try to use
`pip2 install --upgrade motioneye==0.40rc3`
and
`pip install --pre motioneye --upgrade`
but it doesn't work

00lex, 2019-03-30:
try

`systemctl stop motioneye && pip install --upgrade setuptools && pip install --pre motioneye --upgrade && systemctl restart motioneye`

vplustute, 2019-04-01:
No, it doesn't work.

`-sh: systemctl: command not found`

ccrisan, 2019-04-01:
@vplustute either you're on motionEyeOS or you're on a system that uses sysvinit instead of systemd.
In the first case, use the latest nightly build. In the second case, use `/etc/init.d/motioneye stop`.

closed 2019-04-01 by ccrisan