motioneye-project/motioneye #1191: 0.40rc4: streaming issue

00lex, 2019-03-18:
no effect on motion detection or the recorded footage. just the stream is broken 

[https://videos2.sendvid.com/98/04/bua3gbvq.mp4?validfrom=1552944753&validto=1552951953&rate=200k&burst=1000k&hash=h913%2F2zkiYGlSKPaRVUNfeQdYNU%3D](https://videos2.sendvid.com/98/04/bua3gbvq.mp4?validfrom=1552944753&validto=1552951953&rate=200k&burst=1000k&hash=h913%2F2zkiYGlSKPaRVUNfeQdYNU%3D)

ccrisan, 2019-03-19:
@00lex 

> no effect on motion detection or the recorded footage

What *effect* are you expecting?

> https://videos2.sendvid.com/98/04/bua3gbvq.mp4?validfrom=1552944753&validto=1552951953&rate=200k&burst=1000k&hash=h913%2F2zkiYGlSKPaRVUNfeQdYNU%3D

The link is unauthorized here.

00lex, 2019-03-19:
https://cloud.hamburg.freifunk.net/f/e65638267e3849c5bfdd/

ccrisan, 2019-03-19:
@00lex problems like that are often fixed by choosing TCP over UDP when adding the camera. If that still doesn't fix the problem, please run Motion independently of motionEye and report a Motion issue, following [these instructions](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue).

00lex, 2019-03-19:
TCP/UDP doesn't fix it. I make a downgrade to rc3. maybe on Sunday a can make a report

ccrisan, 2019-03-19:
@00lex if downgrading motionEye has an effect, please keep us posted.

00lex, 2019-03-19:
yes. with rc3 everything is ok

MrChris79, 2019-03-20:
I too am having very similar issue with video using 0.40rc4 on my RPi3. However I just came across MotionEye and started playing/testing it. It finds my usb cameras no problem. Been playing with and using Motion by itsself for a while now with no real issues. Maybe if I try rc3 it will fix the video issue or rc5. :)

ccrisan, 2019-03-20:
Fixed in https://github.com/ccrisan/motioneye/commit/114c8c29a383d31f3c680c299f0c8fa66bee1f3d. Please upgrade to 0.40rc5.

closed 2019-03-20 by ccrisan

00lex, 2019-03-20:
works! thx