motioneye-project/motioneye #1192: "Run a command" not working.

WarStormrager, 2019-03-20:
I wrote a simple bash script to turn on a led that I planned to use when motion was detected, and the script worked when using the Linux Terminal, but when trying to run it with the "run a command" function it didn't work.

davidwhenderson3, 2019-03-29:
I wrote a python script for pushover notifications, i.e. yourscriptname.py and placed in /data folder of motionEyeOS. So in the Run a Command input field I put in "python /data/yourscriptname.py" and it worked. I just put in the path at first and it didn't work. I think you have to put in bash then your path to script. i.e. "bash /data/yourscriptname.sh" I'm new to this but that worked for me.

aredon, 2019-09-02:
I had a user issue on mine at one point. You might try using the user daemon in your run a command line.

JamBalaya56562, 2026-11-08:
Closing as answered above: give the full command, with the interpreter if the script is not executable, such as `bash /path/to/script.sh`. The command runs as the user that runs motionEye, which since 0.43.1 is the unprivileged `motion` user, so that user needs access to the script and to the LED, for example through the `gpio` group on a Raspberry Pi.

closed 2026-11-08 by JamBalaya56562