motioneye-project/motioneye #1193: Support for new mask option

richri, 2019-03-22:
Hi,

with motion version 4.1 a new mask option has been introduced: mask_privacy (see here: https://motion-project.github.io/motion_config.html#mask_privacy)

The privacy mask works like the previous existing mask option with the difference, that it completely blacks out the marked parts of the camera image for privacy reasons. The detection is done after the privacy mask filter has been applied.

You already stated in issue #797 that changing the current mask editor is too much effort. May I suggest that instead you add an alternate way for providing the mask definition: by either using your current editor, or by specifiying an existing mask file on the server? This option could be used for both mask files.

So users could be able to create the *.pgm mask file by other means and only reference it via these two options (mask_file and mask_privacy)

Keep up the great work!

Regards, Christian

zagrim, 2019-03-30:
I was just looking into setting up a privacy mask in motionEye, found out mask_privacy option in motion and the lack of support in motionEye, and started wondering how hard would it be to implement support for that in motionEye since there is a mask editor already. For now, I'd myself be fine even with as simple solution as having the currently supported mask being also used as privacy mask. However, having a separate file and an editor GUI for that would of course be better.

I haven't looked into the code base so I have absolutely no clue if it would be hard to implement (and have no prior experience with interfacing motion), but since all the required components would already seem to be there I would assume it shouldn't be too hard. If @ccrisan is too busy with other stuff to implement this I might be willing to help implementing such a feature (it might take some time for me to get into that, though).

zagrim, 2019-04-12:
I worked around the lack of GUI support for privacy masks by using the mask editor to save another mask file, renaming the mask files in filesystem so that both privacy and motion masks can coexist, and adding a line in "Extra Motion Options" (e.g. `mask_privacy mask_1_privacy.pgm`). That's actually quite fine since I don't think I need to adjust privacy masks too often, but I'm still going to look into implementing proper support for this unless someone else gets that done before that.

richri, 2020-02-13:
Are there any plans to add the privacy mask feature to motioneye at all? It would be enormous helpfull if at least the possibility was given to add a privacy mask in the same way you can define a detection mask already.

closed 2020-06-29 by ccrisan

marcusobrien, 2023-01-22:
When this "Privacy Mask" front end widget was added, how does the client camera provide these GUI web options to the web hub server doing the rendering in the browser ? 

Raspberry Pi Zero W 2
I can see the "privacy mask" when my Raspberry Pi Zero W 2 in the web browser when looking at my motioneyeos hub, I can configure it in "Video Settings"

Raspberry Pi Zero W
But in the Raspberry Pi Zero W with the same motioneyeOS installed as the above, no option appears for "Privacy Mask" under the "Video Settings" in the web GUI

Because there is no Privacy Mask setting on the web front end for the Zero W Pi, whenever I change any settings for that Pi oin the browser, the server and client throgu an error. The client camera Pi Zero W says ERROR 'mask'

File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 940, in motion_camera_ui_to_dict
KeyError: 'mask'

PR #1247 (2020-06-29): Add support for privacy mask
This is my proposal for implementing support for privacy masks in motionEye (fixes #1193). The new Privacy Mask related settings are located in Video Device section using similar controls as with motion detection masks. To avoid the need for migrating config dir contents I wanted to keep the original motion detection mask file naming intact even though otherwise I renamed mask stuff with "motion" or "privacy" to avoid confusing the two.