motioneye-project/motioneye #1194: Failed authentication on Update

romankh3, 2019-03-23:
I installed motion Eye on Raspberry Pi and after that tried to update it.
And I've got next:
```
Authentication is required to restart 'motioneye.service'.
Authenticating as: root
```
What kind of credentials do I need to write?
[image]

Best regards, 
Roman B.

ccrisan, 2019-03-25:
Root authentication problems on your system do not constitute a motionEye issue.

closed 2019-03-25 by ccrisan

romankh3, 2019-03-25:
@ccrisan, hello!
I tried to pass my own system credentials and I've got the same result. 
That's why I decided it can be related to MotionEye settings. Am I right?

ccrisan, 2019-03-25:
@romankh3 I don't think you're right. What do your system credentials have to do with motionEye settings?

romankh3, 2019-03-25:
It's default credentials for Raspbian OS. I have installed Motion Eye due to [manual ](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) and after that, I ran update.

ccrisan, 2019-03-25:
@romankh3 I'm still not following you. What do the default Raspbian OS credentials have to do with you installing motionEye?