motioneye-project/motioneye #1195: Please add support for new config options

richri, 2019-03-23:
Hi,

there are a few new config options that make tuning motion detection more effective:

- threshold_maximum (https://motion-project.github.io/motion_config.html#threshold_maximum)
- threshold_tune (https://motion-project.github.io/motion_config.html#threshold_tune)

Especially the first option is very handy when sudden changes due to light etc. would otherrwise create false positives and trigger a motion event.

Could you please consider adding an option to use these settings via the web interface?

Would it otherwise be possible to manually sepcify these options under the field "Extra Motion Options" in the "Video Device" section?

Thanks

Christian

ptryk, 2019-03-25:
hi, I believe you can add these and other config options to the motion daemon into the webinterface via the "extra motion options" (one per line) within the video device section.

ccrisan, 2019-03-26:
@ptryk version 0.41 will probably have them. We're now concentrating on rolling out 0.40 with many changes under the hood.

ccrisan, 2019-05-30:
Fixed via https://github.com/ccrisan/motioneye/pull/1297.

closed 2019-05-30 by ccrisan