motioneye-project/motioneye #1197: Doorbird - Unable to open video device

Dinth, 2019-03-24:
Im struggling to add Doorbird camera feed to MotionEye.
Ifi add it in the following form:
rtsp://10.10.25.10:554/mpeg/media.amp 
and by providing username and password in separate fields.
Im getting a gray screen with "Unable to open stream" message.

ccrisan, 2019-03-24:
Does Motion work standalone with your camera?

Dinth, 2019-03-25:
Im not sure how to test this.
This afternoon i reset Doorbird and readded it to Motioneye and im witnessing different behaviour. Now as soon as i add Doorbird (camera 6) ALL my cameras disappear/error and log shows:
```
 WARNING: mjpg client connection for camera 6 on port 8086 is closed
 WARNING: Connect error on fd 16: ECONNREFUSED
 WARNING: Connect error on fd 16: ECONNREFUSED
 WARNING: Connect error on fd 16: ECONNREFUSED
```
Doorbird works fine in VLC using same similar settings (username and password in url, instead of separately in username and password fields).
[image]

ccrisan, 2019-03-26:
@Dinth use the [Report Motion Issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) instructions to run Motion standalone.

xxorde, 2019-04-19:
@Dinth I am using a Doorbird as well and it works great.
Try to use the latest versions, an easy way to do that is to use the Docker container provided by @ccrisan, ccrisan/motioneye:master-amd64.

Dinth, 2019-04-19:
@xxorde thanks for confirming. Have you added Doorbird by using this url:
rtsp://10.10.25.10:554/mpeg/media.amp and credentials in separate fields
or 
rtsp://username:password@10.10.25.10:554/mpeg/media.amp ?

xxorde, 2019-04-19:
@Dinth  In my case the URL is `rtsp://10.0.0.58:8557/mpeg/media.amp`.

The API documentation (https://www.doorbird.com/downloads/api_lan.pdf) says:

> Returns a MPEG4 H.264 live video stream with the default resolution and compression as defined in the system configuration. Uses RTSP on 554 and the RTSP-over-HTTP protocol on port 8557 of DoorBird and BirdGuard devices
 
```
rtsp://<device-ip>:<device-rtsp-port>/mpeg/
rtsp://<device-ip>:8557/mpeg/media.amp
rtsp://<device-ip>/mpeg/media.amp
```

Its possible to stuff the credentials in the URL as well, but I personally do not like that ;)
I would recommend to create a dedicated user (on the Doorbird) for motion with only the needed permissions.
You can verify if the URL is correct by opening it in VLC.

Dinth, 2019-04-19:
Oh, maybe it needs RTSP-over-HTTP instead of RTSP. I will try this out, thank you!

xxorde, 2019-04-23:
@Dinth did it work?

OberfragGER, 2019-10-24:
> @Dinth In my case the URL is `rtsp://10.0.0.58:8557/mpeg/media.amp`.
> 
> The API documentation (https://www.doorbird.com/downloads/api_lan.pdf) says:
> 
> > Returns a MPEG4 H.264 live video stream with the default resolution and compression as defined in the system configuration. Uses RTSP on 554 and the RTSP-over-HTTP protocol on port 8557 of DoorBird and BirdGuard devices
> 
> ```
> rtsp://<device-ip>:<device-rtsp-port>/mpeg/media.amp
> rtsp://<device-ip>:8557/mpeg/media.amp
> rtsp://<device-ip>/mpeg/media.amp
> ```
> 
> Its possible to stuff the credentials in the URL as well, but I personally do not like that ;)
> I would recommend to create a dedicated user (on the Doorbird) for motion with only the needed permissions.
> You can verify if the URL is correct by opening it in VLC.

Hi,

i confirm your settings! But it was working after a restart of my raspberry... suddenly.
Somehow i blocked my requests: VLC was saying: live555 error: SETUP of'video/H264' failed 429 Too Many Requests
So now- its working.

Camera Device: rtsp://192.168.178.50/mpeg/media.amp
Camera Type: Network Camera
Video Resolution: 1280*720
Streaming Port: 8557

Dinth, 2019-10-25:
It works for me now too. Thank you all.

closed 2019-10-25 by Dinth

pdecat, 2019-11-09:
I had the same issue with `429 Too Many Requests` errors after I first setup my doorbird camera in motioneye in the context of home-assistant using https://github.com/hassio-addons/addon-motioneye.

Stopping and re-starting the motioneye container resolved the issue without changing configuration.

pdecat, 2019-11-09:
Ok, here are the steps to reproduce the issue:
- in motioneye UI, open the drop down menu and click `add camera...`,
- for `Camera Type`, select `Network Camera`
- enter the doorbird rtsp URL in the `URL` field, e.g. rtsp://192.168.1.65/mpeg/media.amp
- as soon as the field looses focus, motioneye scans the URL to detect if the camera is indeed an RTSP camera. The doorbird camera now blocks requests coming from the same client IP address.

Work-around: fill the `Username` and `Password` fields before the `URL` field.

zagrim, 2019-11-09:
I think it might be a good idea to swap the order of credentials and the URL in the UI so that the user would naturally first enter credentials so that the device scan wouldn't go unauthenticated. How do you think of that, @ccrisan?