motioneye-project/motioneye #1201: Streaming Port Change Kills Stream

Spankythemusical, 2019-03-25:
motionEye Version | 0.40rc5
Motion Version | 4.2.2
OS Version | Raspbian 9.8
RPi ZeroW

Changing the Streaming Port from 8081 to any other number (even 8082) results in a grey screen showing "Unable to Open Video Device".  As soon as I change the port back, the stream works again.

ccrisan, 2019-03-26:
I can't seem to be able to reproduce the problem. Can you please attach the resulting configuration files (`motion.conf` and `camera-1.conf`) after applying the problematic setting (remember to remove sensitive data!)?

Spankythemusical, 2019-03-27:
I'm going to go ahead and close this out.  I did an upgrade, as well as a full reformat, on 2 different Pi Zero Ws, all with the same issues.  Found out that I had to manually change /etc/motion/motion.conf in order for any of the options to take effect.  All this oddly left me without motion detection.  Finally I decided to just go with MotionEyeOS, which is working as expected.  Chances are I did something slightly wrong during installation, but couldn't figure it out for the life of me.

closed 2019-03-27 by Spankythemusical