motioneye-project/motioneye #1202: Update Raspbian 9.8 motioneye problem

NikonD7200, 2019-03-29:
motioneye stop to work after raspbian update
motion.log
_motion: error while loading shared libraries: libxãb-drh3.{o>0: cannot open shared object file: No such file or directory_

motion.conf
_# @enabled on
# @show_advanced on
# @normal_password xxxxxxxxx
# @admin_username xxxx
# @admin_password xxxxxxxxxx
# @normal_username xxxx

webcontrol_html_output on
webcontrol_port 7999
setup_mode off
webcontrol_localhost on

thread thread-1.conf
thread thread-3.conf
thread thread-2.conf

webcontrol_parms 2

control_html_output on
control_localhost on
control_port 7999_

bob-lee, 2019-03-30:
What versions of motion and motioneye were you running before upgrading to `9.8`?
I upgraded Raspbian to `9.8` and motioneye `0.40rc5` with motion `4.2.2` seems keep working. You can try reinstalling motioneye:
```
sudo pip uninstall motioneye
sudo pip install motioneye==0.40.rc5
```

NikonD7200, 2019-03-30:
I have motioneye 0.39 uninstall dont work ... i try again

NikonD7200, 2019-03-30:
sudo pip uninstall motioneye = Instruction no allowed

ccrisan, 2019-03-31:
` libxãb-drh3.{o>0` - looks like a filesystem corruption. Definitely not a motionEye issue.

closed 2019-03-31 by ccrisan