motioneye-project/motioneye #1205: MotionEyeOS Login Modification

kish1212, 2019-03-30:
I am actually doing a project where i need to implement a two factor authentication in order to access my camera.
I have installed MotionEyeOS and its running perfectly.
I want to bypass the login page of the MotionEyeOS which is displayed after you enter your IP address.
I have let everything the username and password as default and toggle the Remember Me option, so i don't get the login page.
But now i have already created my webpage in asp.net where where the 2 factor authentication runs perfectly.
Is there a way i can use my asp.net login page to enter the required credentials and then get the MotionEyeOS main page where the live camera is played?
or is there any way where i can embed the live camera in my asp.net page, after i completed the 2 factor authentication process?

So, typing the IP address in the Browser bar should not allow access to the MotionEyeOS.

or is there another way to access the camera without using the IP address?

ccrisan, 2019-04-01:
You can either use one of the URLs that you'll find under *Video Streaming* or you can implement yourself the signature algorithm from [computeSignature](https://github.com/ccrisan/motioneye/blob/dev/motioneye/static/js/main.js#L346) and [addAuthParams](https://github.com/ccrisan/motioneye/blob/dev/motioneye/static/js/main.js#L366).

closed 2019-04-01 by ccrisan

kish1212, 2019-04-10:
Hello ccrisan,
Thanks for your prompt response, i am already using the Video Streaming URLs, now i need to disable access to the motionEyeOS from the IP address of any browser as i will be using only the URLs to view live stream video. So i do not want anyone to be able to access the MotionEyeOS camera by typing the IP address in the browser. So is there a way i can block access to the MotionEyeOS but i should still be able to access the live stream via the Video Stream URLs. Thanks

ccrisan, 2019-04-12:
I recommend using [Nginx](https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx) as a reverse proxy to block access to individual paths.

kish1212, 2019-04-12:
Thank you sir,
 I will google it. I am a newbie and not acquainted with Nginx.  if ever you have any links/guide or tutorials that can help me in achieving this, it would greatly help.
Thanks for your support