motioneye-project/motioneye #1208: Python3 support

ualex73, 2019-04-02:
Currently motionEye only supports Python2, official support for this version stops on 2020.

I did a quick attempt to port motionEye to Python3 with 'six' (then the code is Python2 and 3 compatible) and the parts I have tested, are working fine with motionEye 0.40rc5. Only it requires "tornado 4.5.3", newer version don't work out-of-the-box (motionEye needs new, but incompatible code with Python2).

This week I will create a pull request with the 'six' code, hopefully it will be merged into 0.40rc6.

ualex73, 2019-04-04:
The pull request is created :-)

https://github.com/ccrisan/motioneye/pull/1215

ualex73, 2019-04-20:
Ok, I have created another branch called 'python3', which covers 90% compatibility. Tornado 5.x/6.x use Async, which still give some issues in error situations.

ualex73, 2019-04-26:
The python3 branch has been updated, all asyncio/tornado errors have been resolved.

ccrisan, 2019-05-30:
@ualex73 version 0.40 has been out for some time now and the small remaining bugs and suggestions will be included in 0.41. That being said, I think it's time to consider making the switch to Python 3 in time for 0.50.

At a glance, your code looks fine, but I'll have to run some serious tests (including performance tests) before merging your work.

And by the way, thanks for your work!

ualex73, 2019-06-04:
Thanks :-) If I help anywhere else with this, please let me know.

popoviciri, 2019-08-31:
Hi @ualex73, I just tried your python3 branch on the latest dev of motioneye. I have not been able to access all settings in the frontend. Missing all the "advanced setup" items and others.
It compiles well in buildroot (motioneyeos) and on a linux box as well but with the same result.
I can set everything via the conf files, but the frontend is useless. These are all settings available for user 'admin':
[image]
Care to have a look at your branch and try it with the latest motioneye?
I'm at a loss with python2 to 3 conversion. My skills go as far as 2to3 goes :).
Cheers!

popoviciri, 2019-09-01:
Hi @ualex73, nevermind. I tried your fork directly and got the same result. That means is something on my end.

WarmChocolateCake, 2019-12-10:
Did this get any further?
I've just updated my motioneye system and it has reminded me that Python 2 has very little support remaining.

ualex73, 2019-12-10:
At this moment the PR is stale, because it is waiting Calin to start a branch which support v3.

kleini, 2019-12-11:
How much time will it take to port successfully to Python 3? I am planing to build the Docker images based on Python 3.

ualex73, 2019-12-11:
Shouldn't be too much work, but my 0.40rc port needs to merged with the changes of 0.41.

ccrisan, 2019-12-15:
@ualex73 has been merged into `python3` branch. https://github.com/ccrisan/motioneye/pull/1572 is a work-in-progress PR where work on Python3 compatibility should be tracked.

ektus, 2020-11-27:
Almost one year later, what's the status now? Python 2.7 and even 3.5 are out of support. The HDD on my RPi2 died (after 5 years), so I went on to updating it from wheezy to stretch (and maybe buster 'cause 3.6+ isn't available on stretch, tbd), but ran into problems that seem to indicate motioneye is still stuck on 2.7
[code]
pi@ektusRPi2wheezy1 ~ $ meyectl &
[1] 17238
pi@ektusRPi2wheezy1 ~ $ Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 7, in <module>
    from motioneye.meyectl import main
  File "/home/pi/.local/lib/python3.5/site-packages/motioneye/meyectl.py", line 28, in <module>
    import settings
ImportError: No module named 'settings'
[/code]

ualex73, 2020-11-27:
@ektus I did the Python 2 to 3 port, but I stopped with it (I don't see a merge happen anytime soon)

STrRedWolf, 2021-02-10:
Reference  #1572

toomyem, 2021-10-01:
Any work regarding porting to python3 is in progress?

yaturner, 2021-11-26:
I am a long time user/fan of motioneye and would like to offer my services in helping with the porting to python3, is anyone currently working on this?

starbasessd, 2021-11-26:
There seems to be some issues with certain functions that exist in 2.7 but not in the 3.x versions. This causes a lot of people to 'give up'. It's a shame that the people killing 2.7 in favor of 3.x didn't put more effort into making sure they could safely convert existing functionality, which probably would have allowed 2.7 to be shut down a lot closer to the 2014 EOL than the 2020 EOL...

nullr0ute, 2021-11-26:
Has anyone documented the problematic functions somewhere?

starbasessd, 2021-11-26:
There are utilities that will tell you what's not valid in 3.x if you run the .py scripts through them. I might consider that a project unto itself. I am not a programmer (just a bash/sh/python/dos script kiddie) that does Enterprise/3rd level Help Desk for a living, and doing it here for my version of pay back to the community.

MichaIng, 2022-03-19:
On the way, updated install instructions for testing the Python 3 compatible dev branch are on the way: https://github.com/motioneye-project/motioneye/blob/fcf1b1f/README.md#installation

closed 2022-03-19 by MichaIng