motioneye-project/motioneye #1213: Overriding config values

richri, 2019-04-03:
Hi,

currently I am running a camera on a resolution of 1280x720 pixels, which is 921.600 pixels for the entire image. I now want to set a threshold of, let's say, 750 pixels. which is below the smallest possible value of 0.1% which I could possibly choose via the threshold slider.

My question is: if I now specify an extra motion option as follows: "threshold 750", would this value be persistent and override any value that the slider for threshold might set when I save the config again at a later point?

Thanks, Christian

bkiziuk, 2019-04-04:
According to motion's documentation on [configuration files](https://motion-project.github.io/motion_config.html):

> So always call the camera config files in the end of the motion.conf file. If you define options in motion.conf AFTER the camera file calls, the same options in the camera files will never be used. So always put the camera file call at the end of motion.conf.

So, since Extra Motion Options are attached to the bottom of the config file, they should overrule any previously defined options.

richri, 2019-04-06:
Thnaks for your answe, bkiziuk. However this does not really help me:

In my setup I have several cameras with different settings. So playing a config value in motion.conf rather than camera-1.conf would make it a global config option that is applied to _all_ cameras, which is not what I want.

I just tried adding the option "threshold 750" in the field "Extra Motion Options" under the section "Video Device" of the motioneye GUI. As I said, the camera is running at a resolution of 1280x720. So my config value of 750 is smaller than 0.1% of the total number of pixels (921.600) which I could set via the slider "Frame Change Threshold" under the section "Motion detection".

If I do so, the value seems to be stored in the config file camera-1.conf as expected. However, as soon as I have pressed APPLY and when the GUI is reloaded, I get the error message that an error occurred and that a reload is recommended. This error message persissts even if a restart the motioneye service via Linux shell. Only when I increase the threshold config option in the file back to 921, i.e. to the smallest possible value that the slider could create, the error disappears.

So it seems that even overriding the config value by manually specifying a value for the option threshold, is not option.

I would consider this a bug - since the motioneye gui forces me to configure config options at a bigger value that I need to in my setup.

zagrim, 2019-06-13:
I wonder if it might actually be a good idea to change the scale of the threshold slider... Maybe logarithmic scale would not be good, either, but having some sort of non-linear scale would allow more accurate tuning in the low end without losing any significant accuracy in the high end. I mean, it makes a huge difference on say, 1280x720 resolution, whether threshold is 500 or 1500, but I think it's not that significant whether it is 10.000 or 11.000 (or 150.000 or 151.000 if someone needs to go that high).

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #697, which asks for the same: setting the frame change threshold below what the percent slider allows, e.g. in pixels or with finer steps at the low end. Please follow #697 for it.

closed 2026-11-10 by JamBalaya56562