motioneye-project/motioneye #1214: ONVIF cam support

denisjoshua, 2019-04-04:
Ho there all 
I try to connect an ONFIV cam via Home Assistant docker instalation of Motioneye
Is there an ONFIV support yet please ? 

I try also the RTSP url but I always take an error 404

Thanks all 
Denis

ualex73, 2019-04-04:
ONVIF normally makes it easier to use the camera, under the hood the RSTP URL/stream is used.

motionEye doesn't have ONVIF support, but when my python3 pull request has been included, I will make some code to detect ONVIF camera's (like ZoneMinder, Shinobi, BlueIris,etc have). I am also lazy to figure out the RTSP URL when i add my camera ;-)

denisjoshua, 2019-04-04:
I have try a lot of url type ... for example a link like this one, 
that work on VLC is  rtsp://user:pass@192.168.1.110:554/1/h264major
But in motioneye is not work :-( 
I have a floureon cam
Thanks again 
Denis

ualex73, 2019-04-04:
You can try to increase the motion logging, and check the /var/log/motion.log logfile? Maybe that gives some information why it isn't working. If VLC can open it, in theory motion should work too (motioneye is using motion for the rtsp stream).

denisjoshua, 2019-04-04:
I don't know where is tha log... 
Under /var/log I have only nginx. 
This is a docker installation under Home Assistant (Hassio)
:-( 

Thanks for the help

ualex73, 2019-04-06:
Ok, if it is under Hassio it could be different, but when I check 'https://github.com/hassio-addons/addon-motioneye/blob/master/motioneye/rootfs/etc/motioneye/motioneye.conf' ... It seems to be also under /var/log?

denisjoshua, 2019-04-06:
This is what I have on my motioneye docker: 
`^[[Aroot@a0d7b954-motioneye:/$ ls -la /var/log
total 20
drwxr-xr-x    1 root     root          4096 Feb  4 22:03 .
drwxr-xr-x    1 root     root          4096 Feb  4 22:03 ..
drwxr-xr-x    1 root     root          4096 Apr  4 09:35 nginx
root@a0d7b954-motioneye:/$`
Maybe I'll ask on home assistant forum 
Thanks for the help 
Denis

xing-zheng, 2020-08-20:
> I have try a lot of url type ... for example a link like this one,
> that work on VLC is rtsp://user:pass@192.168.1.110:554/1/h264major
> But in motioneye is not work :-(
> I have a floureon cam
> Thanks again
> Denis

Kindly try to use [Onvif device manager](https://sourceforge.net/projects/onvifdm/) which can be used for getting the rtsp url.

JJSlabbert, 2021-03-12:
If you need ONVIF Pan Tilt support, check this out. 
[](https://www.instructables.com/MOTIONEYE-NVR-PANTILT/)

PeterLeunis, 2022-01-25:
I connected a network camera (CCTronics) in Motioneye with:
URL: rtsp://ipaddress:554/1
User:
Password:

nolantomy, 2022-03-04:
Thanks @PeterLeunis the /1 at the end was what I was missing to setup my iegeek cam.

iurycarlos, 2023-04-15:
any of you had success to use motion detection from the onvif device???

GSVNoFixedabode, 2023-08-19:
Tried this myself and motioneye can't find the device.  The address rtsp://{ip address}:554/onvif1 works perfectly fine if I use VLC, but not within Motioneye. Note, leaving user & password empty.

Dmitrii4444, 2024-09-03:
Hi! I found the solution. All I had to do was to set the settings in UDP in motion eye as on the screenshot
[image]

JamBalaya56562, 2026-11-12:
motionEye adds ONVIF cameras through their RTSP stream; it does not discover or control them over ONVIF itself. As found above, the URL is camera specific (e.g. `rtsp://<ip>:554/onvif1`), and some cameras only work over UDP: after testing the URL in the Add Camera dialog, choose "RTSP/UDP Camera" instead of "RTSP/TCP Camera". Closing as answered.

closed 2026-11-12 by JamBalaya56562