motioneye-project/motioneye #1216: device busy

bmili, 2019-04-04:
Thera is a problem with V4L2 USB Camera. It worked fine , but after I reinstalled motioneyeOs and restored settings, camera is not working. I'm running motioneyeOS on raspberry PI3 B+, Sandisk ultra 32Gb, camera cable is 7m long. Is this a HW problem or could help changing motioneyeOS setting?

Boris
dmesg.log

[257414.358775] bcm2835-v4l2: vidioc_s_fmt_vid_cap device busy
[257414.359947] bcm2835-v4l2: vidioc_s_fmt_vid_cap device busy
[257423.366599] bcm2835-v4l2: vidioc_s_fmt_vid_cap device busy
[257423.368831] bcm2835-v4l2: vidioc_s_fmt_vid_cap device busy
[257424.361326] bcm2835-v4l2: vidioc_s_fmt_vid_cap device busy
[257424.365704] bcm2835-v4l2: vidioc_s_fmt_vid_cap device busy

[root@motioneye log]# tail motion.log
[1:nc5] [DBG] [NET] netcam_read_first_header: Received first header ('Content-length: 33042')
[1:nc5] [DBG] [NET] netcam_check_content_length: Content-Length 33042
[1:nc5] [NTC] [NET] netcam_read_first_header: Content-length present
[1:nc5] [DBG] [NET] netcam_read_first_header: Received first header ('')
[1:ml1] [DBG] [NET] netcam_proc_jpeg: processing jpeg image - content length 33108
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [INF] [NET] netcam_proc_jpeg: return code 6
[1:ml1] [DBG] [NET] netcam_proc_jpeg: processing jpeg image - content length 33108
[1:ml1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1] [INF] [NET] netcam_proc_jpeg: return code 6

motioneye.log
2019-04-04 13:12:41: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&motion_camera_id=4&_signature=d8cf7552e79fd3f9fbad5dd6fb04b96f48868058 (127.0.0.1) 7.06ms
2019-04-04 13:12:41: [motioneye]    DEBUG: received relayed event stop for motion camera id 3 (camera id 3)
2019-04-04 13:12:41: [motioneye]    DEBUG: clearing motion detected for camera with id 3
2019-04-04 13:12:41: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&motion_camera_id=3&_signature=f52598a6373ef8200f48206d3ea39495d2b77666 (127.0.0.1) 7.78ms
2019-04-04 13:12:44: [motioneye]    ERROR: mjpg client timed out receiving data for camera 2 on port 8082
2019-04-04 13:12:45: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 1.72ms
2019-04-04 13:12:50: [motioneye]    DEBUG: received relayed event stop for motion camera id 3 (camera id 3)
2019-04-04 13:12:50: [motioneye]    DEBUG: clearing motion detected for camera with id 3
2019-04-04 13:12:50: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&motion_camera_id=3&_signature=f52598a6373ef8200f48206d3ea39495d2b77666 (127.0.0.1) 7.54ms
2019-04-04 13:12:50: [motioneye]    DEBUG: ignoring event for unknown motion camera id 4
2019-04-04 13:12:50: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&motion_camera_id=4&_signature=d8cf7552e79fd3f9fbad5dd6fb04b96f48868058 (127.0.0.1) 4.66ms
2019-04-04 13:12:50: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 1.82ms

bmili, 2019-04-09:
USB camera works on windows but on RPI  failed to open. It  is model ELP USB with Camera 2.1mm Lens 1080p Hd 

Camera [4:ml4:Camera3] [NTC] [VID] v4l2_device_open: Using videodevice /dev/v4l/by-id/usb-Vimicro_Vimicro_USB_2.0_PC_Camera__Venus_-video-index0 and input -1
[1:ml1] [INF] [NET] netcam_connect: re-using socket 10 since keepalive is set.
[4:ml4:Camera3] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Vimicro_Vimicro_USB_2.0_PC_Camera__Venus_-video-index0: No such file or directory
[4:ml4:Camera3] [ERR] [VID] vid_start: V4L2 device failed to open
[4:ml4:Camera3] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[4:ml4:Camera3] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[4:ml4:Camera3] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[3:ml3:Camera2] [NTC] [ALL] motion_init: Camera 3 started: motion detection Enabled
[3:ml3:Camera2] [NTC] [VID] vid_start: Opening V4L2 device
[3:ml3:Camera2] [NTC] [VID] v4l2_device_open: Using videodevice /dev/video0 and input -1
[3:ml3:Camera2] [DBG] [VID] v4l2_device_capability: ------------------------
[3:ml3:Camera2] [DBG] [VID] v4l2_device_capability: cap.driver: "bm2835 mmal"

jasaw, 2019-04-09:
From your logs:
>Failed to open video device /dev/v4l/by-id/usb-Vimicro_Vimicro_USB_2.0_PC_Camera__Venus_-video-index0: No such file or directory

Make sure that path exists.

closed 2019-04-12 by bmili