motioneye-project/motioneye #1217: When make_message::list_media times out, the sent email ambiguously lacks attachments for no apparent reason

vcaputo, 2019-04-05:
For days I've been spuriously receiving motion notications lacking any attachments, without any time to debug the cause until now.

It appears I've accumulated enough photos to occasionally exceed the default timeout of 10-seconds running on this little Pi Zero, showing this in the logs:

```
2019-04-05 01:21:55: [sendmail]    ERROR: timeout waiting for the media listing process to finish
```
It would be helpful if the notifications included a blurb about the listing process timing out when this occurs, perhaps with a suggestion to go clear out some archives.

I'm also a little confused as to why all of the directories are being scanned to find the recent files for attachment, but that's a separate issue.  It seems like the directories should be used for coarsely limiting the search at least, but I didn't scrutinize the code, maybe I have something misconfigured.

ualex73, 2019-04-05:
You can increase the timeout for the time beeing, this can be done via motioneye.conf, with the following parameters:
# timeout in seconds to wait for media files list, when sending emails
list_media_timeout_email 10

(default is 10, so increase it to 20?)

I don't know the exact code either, but it is a possible limitation of the OS/python. You can't retrieve the 'latest' file, but you need to go through the list to find it out yourself.

vcaputo, 2019-04-05:
@ualex73 Thanks for your response.

I've already increased the timeout to paper over the poor scaling with large archives.

The main thing I wanted to document here was the lack of context in the notification when a timeout occurs.  It would have saved a lot of time and confusion had the notifications said something to the effect of "Attachments unavailable: media list timeout" etc.

As far as the performance of the media listing and it inevitably timing out as more photos accumulate, it's not a language problem it's an implementation problem, but that's orthogonal to this specific issue which is more documentation/UX.

ualex73, 2019-04-06:
@vcaputo I am agreeing with you it isn't 'nice' if such error isn't propagated in the email notification. When/if my Python3 compatibility pull request is merged, I will have a look into the code that we can add this to the email notification.

free-cup-holder-exe, 2026-11-08:
Two changes on `dev` take care of most of this; both are in 0.45.0b1 (`pip install --pre motioneye`):

- #3414: the email sender lists only the pictures inside the attached-images time, not the whole day folder. On a Pi 4 with 57,000 files the listing went from 63 s to 1.5 s, so the default 10 s timeout is rarely reached any more.
- #3413: if the listing still times out, the email goes out without pictures and the log says why, naming `list_media_timeout_email`. Before, the notification was dropped.

The note inside the email you asked for isn't there; the explanation is in the log only.

free-cup-holder-exe, 2026-11-08:
Closing as completed; the line inside the email isn't there, so please reopen if you still want that, or if emails arrive without pictures on 0.45.0b1.

closed 2026-11-08 by free-cup-holder-exe