motioneye-project/motioneye #1222: Timestamp issue

tbartnig, 2019-04-07:
I'm running on docker and have the latest master-amd64 running. Last year I had an issue with the time incorrect displayed , and this was fixed by #1001 

Now in the Netherlands we have summer time from 28-03 and this issue started to appear again.
The timezone in motioneye is two hours back in time.

Any idea how to fix this? In #1001 was mentioned this would be fixed in the next release.
I have this issue too on  timestamps-amd64

ualex73, 2019-04-08:
Where do you see the wrong time?

I am also from the Netherlands, and the video recorded/live view always show at the right bottom the right time (CEST). This default timestamp on the right, is generated by the "motion" process.

tbartnig, 2019-04-08:
I see it in the right corner, but the wrong timestamp is also mentioned in the history of recorded files.

ualex73, 2019-04-08:
That timestamp seem to be generated by motion, not by motionEye. How does your date/time looks like inside the Docker container? Does a restart help?

tbartnig, 2019-04-09:
Docker host (Rancher) Display time in UTC. I'm located in UTC+2.
Date on the docker container is displayed like: Tue Apr  9 17:30:20 Europe 2019

This is not correct, because it's not taken in to account UTC+2. (should be 19:30)

/etc/localtime is an symlink to  /usr/share/zoneinfo/UTC that does not exist.

kleini, 2019-04-09:
You can only solve it by providing your container the correct zoneinfo file for UTC+2. Get the file from somewhere, store it on the host and bind mount it readonly into the container under /etc/localtime.

closed 2019-04-09 by kleini