motioneye-project/motioneye #1223: URL to enable / disable Motion notifications ?

Oliv59, 2019-04-08:
Hello there,

Slightly different from the other topics i found, i don't want to enable or disable the motion detection, but just the detection notifications (email, and run a command).
Of course, without losing all inside setup of those functionalities, to be able to reactivate them still by a simple URL calling, if that's possible ?
Thanks,
Olivier.

corvy, 2019-08-29:
I wonder the same. I want to be able to enable "away" mode from Home Assistant for instance when there is no more movement in the living-room or the alarm is turned on then I want notifications on the camera to be enabled.

aredon, 2019-09-02:
This is possible through motion itself. You need to turn off the localhost limitation so you can see it from another computer. Then you can navigate to things like camera status or enable/disable motion detection. I've used it in my home detection script which is working extremely well.

https://motion-project.github.io/motion_config.html

Read up on the web control section. Here's an example url for camera 0

http://localhost:8080/0/detection/pause

aredon, 2019-09-02:
You'll have to write your own script for this. You could do a simple apache web page to toggle a file or a database entry to determine on or off.

Then have your end command check before doing anything. Motioneye doesn't provide this functionality. 

I have a home detection script I've been using on mine to turn motion detection on and off (and thus also notifications).

corvy, 2019-09-03:
Yes I ended up using the same way, disable motion on motioneye but keep the camera doing motion detection over MQTT so I could continue to use the camera as a motion detection sensor even when turned off on motioneye.

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #2166, which asks for the same and has the longer discussion: switching only the notifications on and off from outside, keeping their settings.

closed 2026-11-13 by JamBalaya56562