motioneye-project/motioneye #1224: Motion Notification - Run a Command

iTom1192, 2019-04-08:
I am very sorry if a question like this has already been answered, but I have search up and down and haven't found a answer.

Under the "Motion Notification" section I already had my Web Hook URL and working, but I wanted to add a command to play a local MP3 file when motion is detected. So I created the following script, tested that it works via terminal (it does), but when I add "sudo bash newdog.sh" in the Run a Command text box it doesn't run after motion is detected. Thoughts on what I am doing wrong?

#!/bin/bash
ffplay -autoexit Mad_Dog-90sec.mp3

iTom1192, 2019-04-09:
I've also tried having the commands set in the "run a command" text boxes under motion detection, file storage, and even in "Run An End Command" under motion detection  to ether,
sudo /etc/motioneye/newdog.sh
/etc/motioneye/newdog.sh

None of the above commands will run the script, but like I said above I just throw /etc/motioneye/newdog.sh at a terminal and it's run the script and play the mp3 file no problem.

jasaw, 2019-04-09:
Try changing `ffplay` and `Mad_Dog-90sec.mp3` in your script to the full path?
Try writing something to some temp file from the script to debug, something like `echo triggered >> /tmp/debug.txt` before calling ffplay. You should be able to see whether `/tmp/debug.txt` is created.

iTom1192, 2019-04-09:
Smart @jasaw, I feel silly for not thinking about that. I added /tmp/debug.txt to the script, and there was a debug.txt in the tmp folder after the script ran. So now that question is why didn't I hear any audio. Since I am not running the script manually from the terminal window do I have to specify the output setting?

#!/bin/bash
echo triggered >> /tmp/debug.txt
ffplay -autoexit /etc/motioneye/Mad_Dog-90sec.mp3

jasaw, 2019-04-09:
First, find out the full path of ffplay by running this command on your terminal: `which ffplay`
Replace the ffplay in your script with the full path.
Maybe dump the ffplay output to `/tmp/debug.txt` and check its contents to see what the problem is, something like `ffplay -autoexit /etc/motioneye/Mad_Dog-90sec.mp3 >> /tmp/debug.txt`
Temporarily increase verbosity of ffplay if you have to.

iTom1192, 2019-04-10:
After all of that I still can't figure out why I can't hear the audio. Though I just had a thought... would having the script always running in the background always checking to see if motion was detected have the same general effect? So my thought would be to add a line in the script that is always looking for motion detection, but that not running the below lines until it does. My only question now is if that works I have no clue what that command line would look like.

iTom1192, 2019-04-10:
So after some more digging I believe the issue the audio device is being used up by using else; thus being the reason why I can't hear the audio. The script is running fine and the mp3 file play, but it has no output since I believe another users has it locked up. Is there a command that would release the audio device, so that when motioneye runs the script it can output to the audio device?

closed 2019-04-10 by iTom1192

iTom1192, 2019-04-11:
The issue has been resolved, and the solution was very simple. Since both ffplay and vlc open up a gui to run the mp3 motioneye didn't like it. When I change the script to have the audio file play via mpg321 motioneye ran the script and played the mp3 file without issue.

serignembow, 2019-07-11:
Please can you tell the command you finally write inside **command **