motioneye-project/motioneye #1229: Run a command when saving file

Dinth, 2019-04-11:
My media files are being saved to the following location: /media/CCTV/Back_garden_left/
I have set up "run a command" option when saving files to submit a message with file location to MQTT server: 
`mosquitto_pub -h 10.10.1.5 -p 1884 -u user -P pass -t cameras/Back_garden_left/motion_filename -m "%f"`
But for some reason the server gets the folliwing message:
`/media/motioneye_10_10_1_6_cctv_cctv/Back_garden_left/2019-04-11/07-10-37.jpg
`
instead of:
`/media/CCTV/Back_garden_left/2019-04-11/07-10-37.jpg
`
So it seems that %f in "Run a command" replaces "CCTV" with "motioneye_10_10_1_6_cctv_cctv" on my system.

woody4165, 2019-05-23:
Hi @Dinth 

Have you solved?

I'm trying to make same thing but in my MQTT message I get %f instead of the filename.

This is my config

[image]

and this is my python file sala_file.py

`mosquitto_pub -h MYIP -u username -P Password -t topic/Sala/file -m "%f"
`

Any suggestion?

Thanks

Dinth, 2019-05-23:
Hi. I think that you're doing it completely wrong. 
1. Your "sala_file.py" is not a python script at all.
2. You have put %f in the script instead of Command field.

Just insert "mosquitto_pub -h MYIP -u username -P Password -t topic/Sala/file -m "%f"" in your command field **instead** of sala_file.py and see if it works.

woody4165, 2019-05-23:
Thanks!

I'm getting with the correct folder 

`/share/motioneye/Camera5/2019-05-23/20-42-21.jpg
`

I'm using MotionEye as Home Assistant addon and this is the version

[image]

Is it the same you're using?

JamBalaya56562, 2026-11-10:
`%f` is the path where Motion actually wrote the file. With "Network Share" storage motionEye mounts the share itself, at `/media/motioneye_<server>_<share>_<user>`, here `10_10_1_6`, `cctv`, `cctv`, so that is the right path; `/media/CCTV` is a different mount of the same share. Use "Custom Path" storage on `/media/CCTV` if the command should see that path. Closing as answered.

closed 2026-11-10 by JamBalaya56562