motioneye-project/motioneye #1231: Pi camera not detected

bob59, 2019-04-11:
Hi
can anyone help please, I have a Raspberry PI Zero W, with a Zero camera module, I have installed Raspbian, enabled the camera, can take a test picture and checked in terminal, supported 1 detected 1.
I then followed the instructions on Motioneye site to install motioneye on Raspberry Pi, it boots ok, and I can get to the motioneye page, but no camera is detected, I have checked the trouble shooting page, all ok and I have the file /dev/video0.

on the main page of MY it lists Motioneye as version 0.39.3, Motion version None, OS version Raspbian 9.8

Any idea's

thanks for your help

bob-lee, 2019-04-11:
I have the same set of gears as yours, initially you have to add camera by yourself on the page. Camera type would be MMAL.

ccrisan, 2019-04-12:
@bob59 you need to install Motion.

closed 2019-04-12 by ccrisan

bob59, 2019-04-12:
thank you both for your help, 
as part of the install I thought I had installed motion 4.1, which was why I was confused, it was not showing on the motioneye front page, but the check motion -h shows nothing, so thanks again for your help.

Bob59

davidrazmadze, 2020-09-28:
Wow. I had the same issue. motionEye started working after I upgraded from 4.1 to 4.2 for "motion".