motioneye-project/motioneye #1232: Cannot install on Raspberry PI Zero

worikgh, 2019-04-12:
Re https://github.com/ccrisan/motioneye/issues/1225

There is not enough memory for this to be installed on a Raspberry PI Zero.  The same a[pplies to MotiuonEyeOS

ccrisan, 2019-04-12:
@worikgh how does that apply to motionEyeOS?

cedric80, 2019-04-16:
FYI:
Since version 6 (released 2nd April 2019), Pillow almost doubled in size causing cache problems with pip. 
You can disabale cache by:
sudo pip --no-cache-dir install motioneye
This should solve the MemoryError.

Cedric

JamBalaya56562, 2026-11-07:
Closing as answered: the MemoryError came from pip's cache with the larger Pillow 6, and `pip install --no-cache-dir` avoids it. For anyone installing now, the current steps are in the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md).

closed 2026-11-07 by JamBalaya56562