motioneye-project/motioneye #1233: Still Cleanup does not work

gardiol, 2019-04-12:
Hi!
latest motioneye (did a pip install --upgrade), clenup does not work. I have a 1.5T partition for motioneye and it gets filled up. After checking, no video has ever been removed since i installed it (over three months).

In motioneye.conf i have "cleanup_interval 36000" it should clean up every 10 hours i think.

I see this issue in the issues list, but it says "fixed", it's not, at least for me?
It's not a file permissions, files, filders and motioneye runs under the same user (not root) and all is on ext4 filesystem

bob-lee, 2019-04-12:
Cleanup sees separate options called `Preserve Movies` and `Preserve Pictures` on the page.

JamBalaya56562, 2026-11-10:
As answered above, the cleanup removes only what is older than Preserve Movies / Preserve Pictures; with those at Forever nothing is ever deleted. No reply since, so closing. Please open a new issue if files older than that setting stay on current motionEye.

closed 2026-11-10 by JamBalaya56562