motioneye-project/motioneye #1236: Pimoroni pantilt hat integrated into Motioneye?

kirstm, 2019-04-14:
Thanks for a great software, keep up the good work!

This is not an issue, instead it should be viewed as cry for help... Have anyone been able to control the Pimoroni pan tilt hat with Motioneye? Perhaps by using the action buttons? I understand this has been done with other controller boards  but I haven't seen anyone doing this with the Pimoroni pan tilt hat.. yet.. 

Unfortunately my knowledge of python is far to limited to write anything useful so my question is whether anyone have integrated Pimoroni pan tilt hat in Motioneye and is willing to share her/his script?  

Here is some information about the Pimoroni pan tilt hat for those that are interested in it

https://pinout.xyz/pinout/pan_tilt_hat#
http://docs.pimoroni.com/pantilthat/#pan
https://www.waveshare.com/w/upload/b/b9/Pan-Tilt_HAT_user_manual_en.pdf

I tried this script, which seems very nice if you have the right hardware, unfortunate it didn't work for me anyway..
https://github.com/luetzel/motion_eye_servo_action

Thank you for taking your time to read this!

/Kirst

Pi-DD, 2019-07-21:
Looking for the same thing!

Pi-DD, 2019-07-23:
Not sure if this is still a live issue for you, but I have now done this and have the Python scripts that enable you to have buttons for up/down, left/right from the web access to the camera that is mounted on the PanTilt Hat, or from "hub" motioneye implementation that is accessing a remote Motioeye camera on a Pi.

Happy to share, just let me know.

jasaw, 2019-07-23:
Fantastic. Can you please share it? I'm sure it will save someone a lot of time. Thanks.

Pi-DD, 2019-07-24:
These are very simple, make no attempt at smooth motion. They simply move in pre-defined steps that you can modify in the script. I found a step of 10degrees was fine for me.

UP button........................

`#!/usr/bin/env python
#Implements "up" button for MotionEye or MotionEyeOS
#Save this file in /etc/motioneye as up_n (no extension!) where n = camera ID
#Make executable
import time
import pantilthat
#Clear buffer
pantilthat.clear()
#Set pan increment between 1 and 90 degrees
#Change this to suit the number of movement steps you want.
incr = 10
#Get current position in range -90 (fully to up) 90 (fully to down)
a = pantilthat.get_tilt()

if a >= -90+incr:
    a -= incr

else:  #prevent attempt to turn beyond 90deg
    a = -90
	
pantilthat.tilt(a)
time.sleep(2)`

DOWN button....................................

`#!/usr/bin/env python
#Implements "down" button for MotionEye and MotionEyeOS
#Save file in /etc/motioneye as down_n (no extension!) where n = camera ID
#Make executable
import time
import pantilthat
#Clear buffer
pantilthat.clear()
#Set pan increment bewteen 1 and 90 degrees
#Change this to suit the number of movement steps you want.
incr = 10
#Get current position in range -90 (fully to right) 90 (fully to left)
a = pantilthat.get_tilt()

if a <= 90-incr:
    a += incr
else:  #prevent attempt to turn beyond 90deg
    a = 90
	
pantilthat.tilt(a)
time.sleep(2)`

LEFT button .....................................................

`#!/usr/bin/env python
#Implements a "left" button for MotionEye and MotionEyeOS
#Save file in /etc/motioneye/ as left_n (no extension!) where n = camera ID
#Make executable
import time
import pantilthat
#Clear buffer
pantilthat.clear()
#Set pan increment between 1 and 90 degrees
#Change to suit the number of movement steps you want. 
incr = 10
#Get current position in range -90 (fully to right) 90 (fully to left)
a = pantilthat.get_pan()

if a <= 90-incr:
    a += incr
else: 
    a = 90
	
pantilthat.pan(a)
time.sleep(2)`

RIGHT button ...............................................

`#!/usr/bin/env python
#Implements "right" button for MotionEye and MotionEyeOS
#Save file to /etc/motioneye/ as right_n (no extension!) where n = camera ID
#Make executable
import time
import pantilthat
#Clear buffer
pantilthat.clear()
#Set pan increment between 1 and 90 degrees
#Change to suit the number of movement steps you want. 
incr = 10
#Get current position in range -90 (fully to right) 90 (fully to left)
a = pantilthat.get_pan()
if a >= -90+incr:
    a -= incr
else:  #Prevent attempt to turn beyond 90deg
    a = -90
	
pantilthat.pan(a)
time.sleep(2)
`

JamBalaya56562, 2026-11-07:
Closing as answered: the comments above share scripts that drive the Pimoroni pan-tilt HAT. motionEye shows an action button for each executable file named after the action and camera in its configuration folder, such as `up_1` or `left_1`. Thanks for sharing them.

closed 2026-11-07 by JamBalaya56562