motioneye-project/motioneye #1238: Unable to update

devils27, 2019-04-16:
The usual pip..... - - upgrade  and also all of the Manual steps of wget new 4+ package gives result of installed wo errors but when accessing motioneye url it still has old version listed. Even after rebooting rasp pi and restarting via systemctl etc. Any ideas?

ccrisan, 2019-04-16:
Make sure you installed it on python2 (with pip2) and make sure you run it from the correct path.

devils27, 2019-04-16:
Yeah just tried pip2 and again manually installing with same result.  My motioneye page still says motion version 4.1.1 and motioneye version 0.39.3 with rasbian os 9.8. It shows it installing everything, tornado, etc but I did see it still said it was installing motioneye 0.39.3, yet i downloaded and dpkg the new 4.1.1.1

devils27, 2019-04-16:
Oh crap.  I'm so sorry.  I just realized the motioneye 4+ is only pre release. I am updatard

closed 2019-04-16 by ccrisan