motioneye-project/motioneye #1243: Failed to start on Debian Jessie

seabird, 2019-04-17:
Hi everyone,

trying to get a CCTV setup working and so far I only managed motion to work/install. Figured I will now spice it up with motioneye. Followed the debian 8 install instructions without any problems. Wen to ip:8765 empty

Back to the server:
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled)
   Active: failed (Result: exit-code) since Wed 2019-04-17 21:14:40 CEST; 1min 27s ago
  Process: 1461 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 1461 (code=exited, status=1/FAILURE)

Apr 17 21:14:40 server meyectl[1461]: File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, in main
Apr 17 21:14:40 server meyectl[1461]: import server
Apr 17 21:14:40 server meyectl[1461]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 28, in <module>
Apr 17 21:14:40 server meyectl[1461]: from tornado.ioloop import IOLoop
Apr 17 21:14:40 server meyectl[1461]: File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 67
Apr 17 21:14:40 server meyectl[1461]: def fileno(self) -> int:
Apr 17 21:14:40 server meyectl[1461]: ^
Apr 17 21:14:40 server meyectl[1461]: SyntaxError: invalid syntax
Apr 17 21:14:40 server systemd[1]: motioneye.service: main process exited, code=exited, status=1/FAILURE
Apr 17 21:14:40 server systemd[1]: Unit motioneye.service entered failed state.

Any help is welcome

ccrisan, 2019-04-19:
Install tornado 5.1.1

JamBalaya56562, 2026-11-08:
Closing as answered by ccrisan above: this was the Python 2 release with an incompatible Tornado, and installing Tornado 5.1.1 fixed it. Current releases run on Python 3 and install a matching Tornado with pip in a virtual environment, see the [README](https://github.com/motioneye-project/motioneye#installation).

closed 2026-11-08 by JamBalaya56562