motioneye-project/motioneye #1248: 'manual_record' not implemented?

ualex73, 2019-04-20:
I was checking the python code if we could manually trigger recording, like with snapshots. It seems partially implemented in the GUI (but not working).

Is it planned in a future release? Seems to be a cool/nice option to have.

ccrisan, 2019-04-20:
@ualex73 if I remember correctly, the reason why I left it like that (partially implemented) was that there was no ideal way to trigger recording in Motion. I don't know if things have changed, but back then the only way to do that was to force triggering of motion detection as well as enable/configure movie recording if it wasn't already.

ualex73, 2019-04-20:
Maybe, I misunderstand the configuration of motion (https://motion-project.github.io/motion_config.html), but the following seems to start/stop the recording with 4.2.2:

{IP}:{port}/{camid}/action/eventstart Trigger a new event.
{IP}:{port}/{camid}/action/eventend Trigger the end of a event.

MichaIng, 2026-10-30:
Action buttons for toggling motion detection and a motion event manually have been added with #3468. Manually triggering a recording only, without e.g. event notification, is difficult, since `motion` does not support such action, but is blocking the camera device, so a manual FFmpeg call our end cannot capture the video stream either. But starting/stopping a manual motion event, with motion-triggered video recording on, comes close.

closed 2026-10-30 by MichaIng