motioneye-project/motioneye #1250: FTP upload stops after one (or a few) uploads

pcbjonnes, 2019-04-24:
Hi,
Over the past week (possibly after updating raspbian / motioneye, but without making any other changes to the configuration), ftp upload has stopped. Motioneye uploads one photo (sometimes several) right after restart of Pi or motioneye, but then it hangs. Test upload in motioneye also does not work when ftp has stopped. The only thing that works then is to restart.
Everything has worked without problems for over a year, and the ftp server has not changed in any way and works for other services.

Is there a log file that logs this activity and what should I look for? Does anyone have any ideas on what might be wrong?

motionEye Version 0.39.3
Motion Version 4.1.1
OS Version Raspbian 9.8

Thanks

bellpaolo88, 2019-04-28:
Same problem here,

> Is there a log file that logs this activity and what should I look for? Does anyone have any ideas on what might be wrong?

Lookslikerain75, 2019-05-02:
Similar issue updated yesterday upload to Google drive stopped working no error switched to ftp nothing uploads. Version 4.2.2

Lookslikerain75, 2019-05-13:
I had to reinstall and reconfigure from scratch not reusing backup of config now working fine.

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.39.3, and one of you got uploads working again with a fresh install and a new configuration. Upload errors are written to `motioneye.log` (`journalctl -u motioneye`). Please open a new issue with that log if uploads still stop after a few files with a current release.

closed 2026-11-08 by JamBalaya56562