motioneye-project/motioneye #1251: Can't add xiaomi Xiaofang hacked camera

gtrhayrug, 2019-04-25:
Hi, using a rtsp flux (working on domotic softwares as jeedom, domoticz), motioneye is not working and leaves a blank screen... Any idea ?

[img]https://i.imgur.com/yoI15Xk.png[/img]

[img]https://i.imgur.com/Mea8gbO.png[/img]

Trying in UDP and TCP, nothing chance and no video flux.

Videoflux works perfectly in VLC using rtsp protocol.

(using last version on debian9 motioneye).

euphoriachieved, 2019-05-21:
Same issue here, I have V2 and it's just been sitting around collecting dust. At one point it actually connected but didn't stay connected for long and I couldn't replicate it to work at all since then.

guillaumeparis2000, 2019-06-27:
Same issue here, I can see the stream with VLC but it doesn't work with motioneye. I have run motion in standalone mode as explained in the wiki. I add the log and my configuration. The Camera 1 is an USB webcam and it work perfectly.

[motion.log](https://github.com/ccrisan/motioneye/files/3333587/motion.log)
[camera-2.conf](https://github.com/ccrisan/motioneye/files/3333590/camera-2.conf.txt)
[motion.conf](https://github.com/ccrisan/motioneye/files/3333591/motion.conf.txt)

gtrhayrug, 2019-06-30:
I've tried to update motioneye to 0.4x last version, but the screen keeps black with the flux rtsp://192.168.2.72:554/unicast (xiafang flux).

guillaumeparis2000, 2019-07-01:
> Same issue here, I can see the stream with VLC but it doesn't work with motioneye. I have run motion in standalone mode as explained in the wiki. I add the log and my configuration. The Camera 1 is [...]

Finaly I solve the problem using a more powerfull raspberry!

goxofy, 2022-11-07:
same problem...

JamBalaya56562, 2026-11-11:
Closing, as this got no logs from a current release. motionEye hands the RTSP URL to Motion as its `netcam_url`, so a black picture from a stream that VLC plays means Motion failed to open or decode it, and `motion.log` (or the system journal) says why. Please open a new issue with that log if a Xiaofang camera still shows nothing with a current release.

closed 2026-11-11 by JamBalaya56562