motioneye-project/motioneye #1252: Straight Rip of RTSP Stream

eckirchn, 2019-04-27:
I have 2 cameras, both HIK IP models.  Each of these have a Main and a Sub Stream.  I am running MotionEye on a Raspberry PI3 with Raspbian (Stretch) and it is working fine.

I would like to take the High Resolution Streams and simply rip those to the drive, without any transcoding.  I have a script I wrote myself to do this, but It would seem this application should be able to do that, but I can'f figure out how to do it.  If motion is detected on the Sub-Streams, I can go back and pull up the Main Stream video for a more detailed event recording without needing the processing power of a real computer.

My example ffmpeg commands, these automatically create date stamped filenames in 60 minute intervals.

`ffmpeg -hide_banner -rtsp_transport tcp -stimeout 1000000 -i rtsp://user:password@192.168.1.90:554/Streaming/Channels/2 -ss 6 -c:v copy -an -flags +global_header -f segment -reset_timestamps 1 -strftime 1 -segment_time 3600 Cam01_%Y%m%d-%H%M-LR.mkv`

`ffmpeg -hide_banner -rtsp_transport tcp -stimeout 1000000 -i rtsp://user:password@192.168.1.90:554/Streaming/Channels/1 -ss 6 -c:v copy -an -flags +global_header -f segment -reset_timestamps 1 -strftime 1 -segment_time 3600 Cam01_%Y%m%d-%H%M-HR.mkv`

Is this something that can be done?

zagrim, 2019-04-28:
I have also a RPi3 setup with two cameras, and tried to do the same thing. In motion there's `netcam_highres` configuration parameter (https://motion-project.github.io/motion_config.html#netcam_highres) which should do exactly this, and which can be put in Extra Motion Parameters in motionEye GUI, and which to me seemed to do what it should. 

However, as I tried to use 1290x960 10 fps main streams, I ran into severe performance issues because 
1) in my setup I was using remote storage over sshfs which is way from optimal for limited CPU (I'm in process to re-thinking that part), and 
2) also motion (or was it ffmpeg, I don't remember) seemed to draw significant extra CPU, I guess that was related 
2.1) motion still doing de/encode to the high res stream because it is applying privacy mask to it, and 
2.2) the h264_omx being blacklisted in raspbian version of ffmpeg (#1730 explains why) and thus the de/encode step requiring much CPU. 

If it is somehow possible to use the commands you are showing, that should prevent the additional de/encoding from occurring and possibly making it feasible in RPi. That might be possible with motion parameters `on_event_start` (run ffmpeg copying) and `on_event_end` (kill the extra ffmpeg process), but the end event might require some work to prevent it from killing the wrong ffmpeg processes.

Anyway, I'm very much interested in solutions for this, too.

zagrim, 2019-04-28:
If you don't have privacy mask configured in motion, please give `netcam_highres` a try as it might actually work for you just fine.

zagrim, 2019-04-30:
@eckirchn There seems to actually exist also `movie_passthrough` option in motion 4.2 (https://motion-project.github.io/motion_config.html#movie_passthrough) which should make the `netcam_highres` stream completely by-pass motion and recude CPU load that way. Please see #1157 (this issue is pretty much a duplicate of that).

eckirchn, 2019-04-30:
@zagrim will look into that, thank you.

JamBalaya56562, 2026-11-08:
Closing as answered by zagrim above, and as a duplicate of #1157. Motion can record the camera's main stream with `netcam_highres` under Extra Motion Options while it detects motion on the sub-stream, and "Movie Passthrough" in the Movies section writes the camera's own video without transcoding, which saves CPU; text overlays and privacy masks are then not in the movie.

closed 2026-11-08 by JamBalaya56562