motioneye-project/motioneye #1253: Set up an alert when a camera connection loss

REYNAUDNicolas, 2019-04-28:
Hello,
How to configure an alert when the connection with a camera is lost ?
Thank you so much !

azogue, 2019-05-12:
You can manually use the keys **'on_camera_lost'** and **'on_camera_found'**, which are analogous to 'on_event_end' and 'on_event_start', so you can run anything when these events happen.

I use that to mirror the motion detection as motion binary_sensors in HomeAssistant, by pushing states not only with event start/end, but also setting as 'motion ON' when connection is lost for some IP cameras, resetting to 'motion OFF' when stream is recovered.

I do the talking with HomeAssistant with a simple POST with `curl`, using a long-live token made just for motioneye in HA. My `thread-X.conf` file (now named `camera-X.conf`) looks like this:

```conf
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t; curl -X POST -H "Authorization: Bearer AUTH_TOKEN" -H "Content-Type: application/json" -d '{"state": "off", "attributes": {"device_class": "motion", "friendly_name": "Vídeo-Mov. en Salón"}}' http://HA_LOCAL_IP:8123/api/states/binary_sensor.motioncam_salon
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t; curl -X POST -H "Authorization: Bearer AUTH_TOKEN" -H "Content-Type: application/json" -d '{"state": "on", "attributes": {"device_class": "motion", "friendly_name": "Vídeo-Mov. en Salón"}}' http://HA_LOCAL_IP:8123/api/states/binary_sensor.motioncam_salon
on_camera_lost curl -X POST -H "Authorization: Bearer AUTH_TOKEN" -H "Content-Type: application/json" -d '{"state": "on", "attributes": {"device_class": "motion", "friendly_name": "Vídeo-Mov. en Salón"}}' http://HA_LOCAL_IP:8123/api/states/binary_sensor.motioncam_salon
on_camera_found curl -X POST -H "Authorization: Bearer AUTH_TOKEN" -H "Content-Type: application/json" -d '{"state": "off", "attributes": {"device_class": "motion", "friendly_name": "Vídeo-Mov. en Salón"}}' http://HA_LOCAL_IP:8123/api/states/binary_sensor.motioncam_salon
```

**Because these keys cannot be edited with the MotionEye UI, you need to manually edit the camera configuration**, and take into account that MEye could remove these keys when over-writing the cam config from the UI

REYNAUDNicolas, 2019-05-12:
Thanks !
I put this in "Extra Motion Options", I noticed that it added it on file configuration of the camera.

But when the camera is disconnected the image is freeze and the connection is not marked lost.

JamBalaya56562, 2026-11-08:
Closing as answered above: Motion's `on_camera_lost` and `on_camera_found` options, added under Extra Motion Options, run a command when a camera disconnects and comes back. A stream that freezes without disconnecting does not count as lost for Motion, which is why the frozen picture did not trigger it; that part is up to [Motion](https://github.com/Motion-Project/motion).

closed 2026-11-08 by JamBalaya56562