motioneye-project/motioneye #1254: SSL for streaming

flokX, 2019-05-01:
Hi, I know that this feature request was already a few times in the motioneyeos repo, but I think the circumstances changed now.

Motion introduced in the version 4.2 the [stream_tls](https://motion-project.github.io/motion_config.html#stream_tls) function and offers the full solution to secure the streams with SSL.
You bumped motion in the project already to 4.2 (#1172), so perhaps the function is now worth a look.

In some setups like mine it is already fine to secure the streams via HTTPS only because these are accessible from outside, while the web interface is only in my home network. Implementing this feature would be a big security gain.

chrisf4lc0n, 2019-05-08:
Use NGINX or Lighttpd Reverse Proxy.

flokX, 2019-05-09:
I know this solution, but it's easier to configure and maintain if everything is controlled by motioneye.

JamBalaya56562, 2026-11-15:
motionEye reads each camera's stream from motion itself, over plain HTTP on localhost, to show it in the UI, so turning on motion's `stream_tls` would break the preview, and it would also need certificates managed on every device. A reverse proxy with HTTPS in front of motionEye and the stream ports, as suggested above, is the way to secure them from outside. Closing as not planned.

closed 2026-11-15 by JamBalaya56562