motioneye-project/motioneye #1255: Is there a way to enable streaming only when needed ?

SR-G, 2019-05-01:
Hello.
I'm new to MotionEye.
I have two Xiaomi Yi Dome Cameras streaming RTSP and configured under MotionEye (docker container).
Everything works fine but the CPU usage is really huge ... (even with the "pass-through" feature referenced #1157 and that even seems to consume more than a transcodification to a low resolution)

So, as i'm not willing to use the motion detection features, is there a way to :

- have 0 stream conversions being done when there are no viewers (hence, 0 CPU used)
- automatically (re)start the streaming as soon as / only once someone is triggering any kind of URL (snapshot, MJPG, ...) (even if there is a small latency at that time)

jakepals, 2020-02-21:
I sometimes use crontab to toggle the motioneye service for linux. (eg sudo systemctl stop/start motioneye)

first open up crontab from a terminal
**`sudo crontab -e`**

it will ask which editor to modify (nano is simplest to use)

here's my example for crontab (add to the bottom of the document)

**`0 8 * * 0-5 systemctl stop motioneye`
`0 20 * * 0-6 systemctl start motioneye`**

this is saying that,
**on 8am Sunday through Thursday, turn off motioneye
and at 8pm Sundays through Fridays, turn on motioneye** 

change it around to what fits you best, and google how to get more specific on crontabs (at this time, i'm not that great at crontabs, i had to do some googling to get my schedule set up. i even went to some sites that help generate crontab schedules)

hope this helps

JamBalaya56562, 2026-11-10:
Closing, as Motion reads a camera's stream all the time while the camera is enabled, whether anyone watches or not, and motionEye has no on-demand mode for it. What it can do is switch a camera off: with the camera's Enabled switch off, Motion no longer reads it at all, and stopping the service from cron, as suggested above, does the same for every camera.

closed 2026-11-10 by JamBalaya56562