motioneye-project/motioneye #1256: BUG upgrade from 0.39 to 0.40

marcorighi, 2019-05-01:
After the upgrade, the system is misconfigurated and ask for new cam. Please resolve the bug,

ccrisan, 2019-05-02:
That is not how you report a bug. Have you followo the indications in the release notes?

efelon, 2019-05-02:
raspbian stretch (latest upgrades)

Just upgraded successfully from 0.39.4 to 0.40 using pip and the migration script in the [release notes]( https://github.com/ccrisan/motioneye/releases/tag/0.40). I had to make a minor change to the given command because the path to migrateconf.sh was different on my system ("dist-packages" instead of "site-packages"):
```
for file in /etc/motioneye/{motion,thread-*}.conf; do /usr/local/lib/python2.7/dist-packages/motioneye/scripts/migrateconf.sh $file; done
```
Worked out of the box after that.
Thank you very much @ccrisan.

grumpazoid, 2019-05-16:
Many thanks efelon for the dist-packages advice.
I was still getting permission errors on raspian until I initiated under root (sudo su)
Now have my camera vies back!

tharfy, 2019-06-20:
The script worked like a charm except for one of my cameras, that works is udp mode (Konx doorbell).
I had to replace :
`netcam_use_tcp off`
by
`rtsp_uses_tcp off`

JamBalaya56562, 2026-11-07:
Closing as answered, same cause as #1393: the configuration has to be migrated as described in the [0.40 release notes](https://github.com/motioneye-project/motioneye/releases/tag/0.40), and the comments above confirm that the script works.

closed 2026-11-07 by JamBalaya56562