motioneye-project/motioneye #1258: Memory Leak in 0.40

jstride, 2019-05-03:
I've just pulled the latest ccrisan/motioneye:master-amd64 from dockerhub and seem to have a really strange memory leak issue.

When saving a change to an existing camera (I ran the migration script) it takes much longer to save and then the memory usage goes through the roof and I can't connect to the container. I do see the following in the logs:

```
 WARNING: Connect error on fd 16: ECONNREFUSED
 WARNING: Connect error on fd 16: ECONNREFUSED
 WARNING: Connect error on fd 17: ECONNREFUSED
 WARNING: Connect error on fd 18: ECONNREFUSED
 WARNING: Connect error on fd 19: ECONNREFUSED
```

peturdainn, 2020-03-02:
Not sure if I'm seeing the same issue/leak, but after running stable for a day, interacting with it makes memory consumption go up until everything is consumed. No idea what interaction, but it is more than viewing. Probably settings.

I see Python2.7 go consume several GB of RAM at a rate of 100MB/30sec, and it goes way past the limit set for the container. 

Once this happens, even rebooting the server doesn't help, as soon as the container starts it is consuming memory at an alarming rate.

So maybe a config corruption (will investigate)

JamBalaya56562, 2026-11-08:
Closing, as this was with the Python 2 image of 0.40 and got no logs beyond `ECONNREFUSED`. That message means motionEye could not connect to Motion. Please open a new issue with `motion.log` and `motioneye.log` if memory still grows when saving a camera with a current image.

closed 2026-11-08 by JamBalaya56562