motioneye-project/motioneye #1259: netcam_highres  for manual screenshot

aLiEnHeAd13, 2019-05-03:
Hi,
I configured my camera according to 
https://github.com/ccrisan/motioneye/issues/880
with a low-res viewing stream and a hi-res recording stream via netcam_highres.
The motion deteced screenshots and movies are hi-res but manual screenshots are low-res.
In my opinion manual screenshots should use the high resolution as well.

ualex73, 2019-05-03:
This sounds more of a motion request, and not a motionEye request. motionEye just calls the API to trigger a snapshot.

zagrim, 2019-05-04:
@aLiEnHeAd13 are you using `movie_passthrough` together with `netcam_highres`, or just `netcam_highres` alone? Motion docs state quite clearly that with `movie_passthrough` the behaviour should be just the way you've experienced, but the docs don't mention how it behaves without the passthrough option. Anyway, you should ask that from [motion project](https://github.com/Motion-Project/motion/issues).

aLiEnHeAd13, 2019-05-04:
Hi @zagrim the only additional motion option is netcam_highres. No passthrough. OK. I'll have a look at the motion project.

aLiEnHeAd13, 2019-05-06:
created an issue at motion:
[https://github.com/Motion-Project/motion/issues/942](https://github.com/Motion-Project/motion/issues/942)

closed 2019-05-06 by aLiEnHeAd13