motioneye-project/motioneye #1261: Timestamp issues

NginUS, 2019-05-04:
I'm on Linux Mint 18.3 (Ubuntu 16.04 LTS)
motionEye Version | 0.40
Motion Version | 4.2.2

Today I lost all my configuration after running an upgrade script I made:
sudo pip install --upgrade pip && sleep 2 && sudo pip install motioneye --upgrade && sleep 2 && sudo systemctl daemon-reload && sleep 2 && sudo systemctl restart motioneye

After reconfiguring everything I now have a problem where the 'Text Overlay' option isn't respected. I turn it off it stays on. I change the left/right content it stays the same. I change the size it stays small. There's also a bunch of artifacts covering the time where the seconds create constant motion on the image there in the corner. Though that doesn't appear on the recorded images.

Its not a big deal but it's not right so I figured I'd post here.

I'm not sure if my versions are current, or if my update script is doing what needs to be done properly. Would be nice if this is just a version issue with a quick fix, but I have no idea.

Could use a hand. Thanks.

ccrisan, 2019-05-19:
> Today I lost all my configuration after running an upgrade script I made:
sudo pip install --upgrade pip && sleep 2 && sudo pip install motioneye --upgrade && sleep 2 && sudo systemctl daemon-reload && sleep 2 && sudo systemctl restart motioneye

Have you followed the upgrade procedure indicated in the [0.40 release notes](https://github.com/ccrisan/motioneye/releases/tag/0.40)?

> After reconfiguring everything I now have a problem where the 'Text Overlay' option isn't respected. I turn it off it stays on. I change the left/right content it stays the same. I change the size [...]

Can we see some screenshots exemplifying this behavior?

NginUS, 2019-05-19:
> Have you followed the upgrade procedure indicated in the [0.40 release notes](https://github.com/ccrisan/motioneye/releases/tag/0.40)?
No, I did not. Will be aware to check for that next time. Thanks.

> Can we see some screenshots exemplifying this behavior?
The upper text is from the camera itself, and the lower set is from Motioneye. In this case the 'Text Overlay' toggle is in the off position. Although when it's 'on', the scale & content remain the same as is pictured here.
[image]

Dlay2017, 2020-11-20:
> No, I did not. Will be aware to check for that next time. Thanks.
>
>
> The upper text is from the camera itself, and the lower set is from Motioneye. In this case the 'Text Overlay' toggle is in the [...]

Try to upgrade motion to version 4.3.2, for ex from here: [https://github.com/Motion-Project/motion/releases](url)
Worked for me

JamBalaya56562, 2026-11-08:
Closing as answered by Dlay2017: the text overlay is drawn by Motion, and upgrading Motion from 4.2.2 to 4.3.2 fixed it for them. Current releases work with Motion 4.x from the OS packages. Please open a new issue if the overlay still ignores its settings.

closed 2026-11-08 by JamBalaya56562