motioneye-project/motioneye #1262: Errors in log and not detecting movement

ivallesp, 2019-05-05:
Hi! 

I managed to install motioneye in my raspberry pi 3 and the server seems to be running and does not show  any big problem. I have been trying to configure the notifications by mail and doing some captures and there's where I found a problem. Apparently the GUI does not show any issue, but when I take a look at the log, I see some errors (see below). I tried everything I found on the internet but no success: increasing RAM, setting LD_PRELOAD environment, etc.

Can somebody point me to potential issues? Thanks!

```pi@raspberrypi:~ $ journalctl -u motioneye > motioneye.log
pi@raspberrypi:~ $ cat motioneye.log 
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Sun 2019-05-05 00:48:25 BST. --
May 05 00:47:24 raspberrypi systemd[1]: Started motionEye Server.
May 05 00:47:29 raspberrypi meyectl[277]:     INFO: hello! this is motionEye server 0.39.3
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [WRN] [ALL] thread config option deprecated use camera
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 05 00:47:33 raspberrypi motion[510]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 05 00:47:33 raspberrypi motion[510]: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
May 05 00:47:33 raspberrypi motion[510]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:47:33 raspberrypi motion[510]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:47:33 raspberrypi motion[510]: [1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
May 05 00:47:33 raspberrypi motion[510]: [1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
May 05 00:47:35 raspberrypi meyectl[277]:     INFO: cleanup started
May 05 00:47:35 raspberrypi meyectl[277]:     INFO: wsswitch started
May 05 00:47:36 raspberrypi meyectl[277]:     INFO: tasks started
May 05 00:47:36 raspberrypi meyectl[277]:     INFO: mjpg client garbage collector started
May 05 00:47:36 raspberrypi meyectl[277]:     INFO: server started
May 05 00:47:43 raspberrypi motion[510]: [1:ml1] [ERR] [ALL] motion_init: Error capturing first image
May 05 00:47:45 raspberrypi meyectl[277]:    ERROR: motion not running, starting it
May 05 00:47:45 raspberrypi motion[861]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 05 00:47:45 raspberrypi motion[861]: [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
May 05 00:47:45 raspberrypi motion[861]: [0:motion] [WRN] [ALL] thread config option deprecated use camera
May 05 00:47:46 raspberrypi motion[861]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
May 05 00:47:46 raspberrypi motion[861]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
May 05 00:47:46 raspberrypi motion[861]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 05 00:47:46 raspberrypi motion[861]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 05 00:47:46 raspberrypi motion[861]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 05 00:47:46 raspberrypi motion[861]: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
May 05 00:47:46 raspberrypi motion[861]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:47:46 raspberrypi motion[861]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:47:46 raspberrypi motion[861]: [1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
May 05 00:47:46 raspberrypi motion[861]: [1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
May 05 00:48:02 raspberrypi meyectl[277]:  WARNING: mjpg client connection for camera 1 on port 8081 is closed
May 05 00:48:03 raspberrypi motion[861]: [1:ml1] [ERR] [ALL] motion_init: Error capturing first image
May 05 00:48:12 raspberrypi meyectl[277]:    ERROR: motion not running, starting it
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [WRN] [ALL] thread config option deprecated use camera
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 05 00:48:13 raspberrypi motion[865]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 05 00:48:13 raspberrypi motion[865]: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
May 05 00:48:13 raspberrypi motion[865]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:48:13 raspberrypi motion[865]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:48:13 raspberrypi motion[865]: [1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
May 05 00:48:13 raspberrypi motion[865]: [1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
May 05 00:48:23 raspberrypi motion[865]: [1:ml1] [ERR] [ALL] motion_init: Error capturing first image
May 05 00:48:23 raspberrypi motion[865]: [1:ml1] [ERR] [ALL] motion_init: Problem enabling motion-stream server in port 8081: Address already in use
May 05 00:48:24 raspberrypi meyectl[277]:    ERROR: motion not running, starting it
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [WRN] [ALL] thread config option deprecated use camera
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 05 00:48:25 raspberrypi motion[875]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 05 00:48:25 raspberrypi motion[875]: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
May 05 00:48:25 raspberrypi motion[875]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:48:25 raspberrypi motion[875]: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
                                         VIDIOC_S_FMT: : Device or resource busy
May 05 00:48:25 raspberrypi motion[875]: [1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
May 05 00:48:25 raspberrypi motion[875]: [1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.```

ivallesp, 2019-05-11:
Anybody here, please?

MichaIng, 2022-10-09:
@ivallesp 
Is this still an issue? If so, which `motion` version do/did you use?
```sh
motion -h
```
This setting was introduced was motion v4.1, it seems: https://motion-project.github.io/4.1.1/motion_guide.html

closed 2022-10-09 by MichaIng