motioneye-project/motioneye #1263: Root Privilege Error

mitch1138, 2019-05-05:
I have tried to run motioneye as installed (https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) and as a normal user (https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User) but I can't get either to work.  I think I may have not setup the privileges properly.  Hopefully it is just my inexperience.  Here is the system status, do anyone have any guidance.
Thanks

**systemctl status motioneye**
 motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-05-05 11:42:07 EDT; 7s ago
  Process: 893 ExecStart=/usr/local/bin/meyectl startserver -c /home/pi/motioneye/motioneye.conf (code=exited, status=255)
 Main PID: 893 (code=exited, status=255)

May 05 11:42:05 raspberrypi systemd[1]: Started motionEye Server.
May 05 11:42:07 raspberrypi meyectl[893]:     INFO: hello! this is motionEye server 0.39.3
May 05 11:42:07 raspberrypi meyectl[893]: CRITICAL: smb shares require root privileges
May 05 11:42:07 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=255/n/a
May 05 11:42:07 raspberrypi systemd[1]: motioneye.service: Unit entered failed state.
May 05 11:42:07 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.

ccrisan, 2019-05-19:
@mitch1138 the log clearly says what the problem is:

    May 05 11:42:07 raspberrypi meyectl[893]: CRITICAL: smb shares require root privileges

You have configured your motionEye to run as a normal user but you have also enabled SMB shares, which requires root. You can't have both.

closed 2019-05-19 by ccrisan

mitch1138, 2019-08-13:
Thank you for response.  I am still getting the same error, clearly an issue on my end and I have been working on it since May.  I don't have Samba loaded and I don't know how to disable SMB shares.  When I load motioneye on a clean system is works fine, but after reboot I get this error.  Any guidance would be greatly appreciated.

[motioneye error.txt](https://github.com/ccrisan/motioneye/files/3498132/motioneye.error.txt)