motioneye-project/motioneye #1265: Set Timezone in Docker installation

woody4165, 2019-05-07:
Hi

I've followed the instruction on [https://github.com/ccrisan/motioneye/wiki/Install-In-Docker#image-from-docker-hub](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker#image-from-docker-hub) to install Motioneye in Docker, but I get the date in UTC while I am on CET (GMT+2).

I've tried to follow some guides to change Timezone, but with no results.

I'm not even able to install tzdata within the container (I'm using Portainer, accessing the Console to send command within the MotionEye container), but I get this error using command 
`apt-get install tzdata`

```
dpkg: error processing archive /var/cache/apt/archives/tzdata_2017c-0ubuntu0.17.10_all.deb (--unpack):
 unable to make backup link of './usr/share/zoneinfo/UTC' before installing new version: Invalid cross-device link
Errors were encountered while processing:
 /var/cache/apt/archives/tzdata_2017c-0ubuntu0.17.10_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

What can I do to change Timezone within the docker image?

Thanks

ccrisan, 2019-05-19:
@kleini, @Hascki any ideas?

kleini, 2019-05-20:
The guide https://github.com/ccrisan/motioneye/wiki/Install-In-Docker#image-from-docker-hub clearly instructs, to bind the timezone from outside of the Docker, from the host into the containers /etc/localtime file. Every guide telling you to install something inside the container is not useful. There should be no writes inside running Docker containers and that includes installation of packages. Writing files inside Docker containers often results in kernel panics on the host, letting your complete host die.

Did you try to bind the /etc/localtime timezone information from your host into your container? Does your host have a proper timezone information configured?

woody4165, 2019-05-20:
Hi @kleini I've followed exactly what's written in that guide.

I'm on a Proxmox VM and inside the VM the result of date is

`Mon May 20 13:29:48 CEST 2019
`
and this is from the MotionEye container (using the Portainer console)

`Mon May 20 11:32:18 UTC 2019
`
When I've created the container I have done

```
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-amd64
```

Thanks

kleini, 2019-05-20:
@woody4165 Then please check what is behind /etc/localtime on the host. This looks as follows on my systems:

    $ ls -l /etc/localtime 
    lrwxrwxrwx 1 root root 33 Apr 29 10:30 /etc/localtime -> /usr/share/zoneinfo/Europe/Berlin

woody4165, 2019-05-20:
That's the difference!

From the MotionEye container

[image]

From the VM

[image]

kleini, 2019-05-20:
Okay, something is weird here. When you're using the bind mount correctly, /etc/localtime should be a file and not a symlink.
Second, your host has obviously UTC configured and not Europe/Berlin. So your host is on UTC and not CEST or GMT+2.
I have no idea, how to fix that.

kleini, 2019-05-20:
Okay, Docker is somewhat weird! I have the same symlink inside the container.

    # ls -l /etc/localtime 
    lrwxrwxrwx 1 root root 23 May 24  2018 /etc/localtime -> /usr/share/zoneinfo/UTC

But that file is not the UTC timezone. I copied that file from inside the container to outside to verify its contents.

    strings UTC
    TZif2
    	LMT
    CEST
    CEMT
    TZif2
    	LMT
    CEST
    CEMT
    CET-1CEST,M3.5.0,M10.5.0/3

So based on the bind mount Docker seems to copy the file somehow into the container and it copies it to the symlink destination. This is really weird!

kleini, 2019-05-20:
This will require now a lot of research on how Docker implements file bind mounts and how to properly transport the timezone configuration of the host into the container.

woody4165, 2019-05-20:
It's really weird.

I'm using Home Assistant (Hassio) in docker in another VM on the same Proxmox server

When I select an addon (also MotionEye) it will create a docker container and in that container the date and TZ is correct.

In that container there is an environment variable
TZ  Europe/Rome

that is missing in the "normal" MotionEye

kleini, 2019-05-21:
I don't think, using this environment variable is the right way to implement proper timezones inside Docker containers. This requires to add the information about all possible timezones into the container; you need to install the tzdata package. This increases the size of Docker images while the normal aspiration should be to reduce the size of the images. Inside the container you only need one timezone, normally the same as on the host. Therefore using the bind mount of the timezone file of the host is the better solution as it reduces the image size.

zalouma1, 2020-04-02:
I have correctly Prague timezone defined in Raspbian. My `docker-compose` contains:
```
version: "3.5"
services:
  motioneye:
    volumes:
      - /usr/share/zoneinfo/Europe/Prague:/etc/localtime:ro
```

ma-zal, 2020-04-02:
I have correctly Prague timezone defined in Raspbian. My docker-compose contains:
```
version: "3.5"
services:
  motioneye:
    volumes:
      - /usr/share/zoneinfo/Europe/Prague:/etc/localtime:ro
```

Vrumf0ndel, 2022-08-18:
I had also the same problem with wrong timezone in docker container on Unraid.
The timezone is set correctly via `-e TZ="Europe/Berlin"` parameter in docker run command line, but docker container uses "Europe" timezone only (checked via `date +%Z` in shell).

So my solution is to map the zone info with `-v /usr/share/zoneinfo:/usr/share/zoneinfo:ro` to the docker container.
Now it works for me as expected.

MichaIng, 2022-08-18:
Is it the same with the new dev Docker container? https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

Vrumf0ndel, 2022-08-19:
I'm using the Community Applications on Unraid, and the only available docker container for motioneye in this catalog is `https://hub.docker.com/r/malvarez00/docker-motioneye`
The versions are quite old: motionEye Version 0.42, Motion Version 4.0
It looks like this container is no longer maintained. 

Haven't tried docker container from motioneye-project until now, but I will give it a try.

Vrumf0ndel, 2022-08-19:
@MichaIng With newest dev docker container the time zone problem seems to be solved.
Just using docker run parameter `-e TZ="Europe/Berlin"` and the time stamp is shown correct.

JamBalaya56562, 2026-10-31:
The current images (`motioneyeproject/motioneye`) include tzdata, so `-e TZ=Europe/Berlin` works, as does mounting `/etc/localtime` as shown in [docker/README.md](https://github.com/motioneye-project/motioneye/blob/dev/docker/README.md). Checked with `latest` (0.44.0) and `edge`: `date` gives CET. Closing as completed.

closed 2026-10-31 by JamBalaya56562