motioneye-project/motioneye #1266: From 0.40rc5-armhf to 0.40-armhf a lot of logs

mcgyver83, 2019-05-07:
Moving from 0.40rc5-armhf to 0.40-armhf docker image a lot of log lines happear

`DEBUG: creating mjpg client for camera 4
   DEBUG: mjpg client for camera 4 connected on port 9004
   DEBUG: creating mjpg client for camera 3
   DEBUG: mjpg client for camera 3 connected on port 9003
   DEBUG: mjpg client for camera 1 on port 9001 has been idle for 10 seconds, removing it`

And also a lot of lines in motion.log inside the container
`[7:nc7:FirstFloorCam] [INF] [ENC] ffmpeg_avcodec_log: error while decoding MB 73 0
[7:nc7:FirstFloorCam] [INF] [ENC] ffmpeg_avcodec_log: max delay reached. need to consume packet
[7:nc7:FirstFloorCam] [INF] [ENC] ffmpeg_avcodec_log: RTP: missed 178 packets
[0:cn0] [DBG] [STR] webu_parseurl: Sent url: /2/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /2/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: Sent url: /1/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: camid: >2< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2:>< val2:><
[0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /1/detection/status
[0:wu0] [INF] [ALL] webu_clientip: Connection from: 127.0.0.1
[0:cn0] [DBG] [STR] webu_parseurl: camid: >1< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2:>< val2:><
[0:wu0] [INF] [ALL] webu_clientip: Connection from: 127.0.0.1
[0:cn0] [DBG] [STR] webu_parseurl: Sent url: /4/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /4/detection/status`

I tried to set log_level for motioneye and motion but no luck :(

kleini, 2019-05-07:
I have to remove debug logging again.

mcgyver83, 2019-05-07:
In fact the log_level configuration property isn't working.
Do you plan to fix this log_level capability or just to remove debug stuff?

kleini, 2019-05-07:
It is not working because it is overwritten using a command line parameter for motioneye in the CMD instruction of the Dockerfile. I was not able to set up an empty motioneye without this parameter. The empty motioneye did not work then. This would be a larger issue.

closed 2019-05-08 by ccrisan

mcgyver83, 2019-05-16:
I updated the docker image ccrisan/motioneye:master-armhf but I still have a lot of logs in motion.log

`[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: max delay reached. need to consume packet
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: RTP: missed 23 packets
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: corrupted macroblock 36 14 (total_coeff=-1)
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: error while decoding MB 36 14
[0:cn0] [DBG] [STR] webu_parseurl: Sent url: /2/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /2/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: camid: >2< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2:>< val2:><
[0:cn0] [DBG] [STR] webu_parseurl: Sent url: /1/detection/status
[0:wu0] [INF] [ALL] webu_clientip: Connection from: 127.0.0.1
[0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /1/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: camid: >1< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2:>< val2:><
[0:wu0] [INF] [ALL] webu_clientip: Connection from: 127.0.0.1
[0:cn0] [DBG] [STR] webu_parseurl: Sent url: /4/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: Decoded url: /4/detection/status
[0:cn0] [DBG] [STR] webu_parseurl: camid: >4< cmd1: >detection< cmd2: >status< parm1:>< val1:>< parm2:>< val2:><
[0:wu0] [INF] [ALL] webu_clientip: Connection from: 127.0.0.1
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: max delay reached. need to consume packet
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: RTP: missed 14 packets
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: max delay reached. need to consume packet
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: RTP: missed 5 packets
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: Invalid level prefix
[5:nc5:Scala] [INF] [ENC] ffmpeg_avcodec_log: error while decoding MB 29 21`

PR #1267 (2019-05-08): Removed debug option for Docker images
Resolves #1266.