motioneye-project/motioneye #1268: No video streaming from Camera

ewanyande, 2019-05-08:
I have installed motioneye on a PC with ubuntu 16.04lts but when I connect a camera it shows the "no signal sign". The same settings work well on motioneye on Raspberry. Please assist. I have attached a screenshot of the camera settings too.

[image]
[image]

The raspberry uses the following:

- motionEye Version | 0.39.3
- Motion Version | 4.1.1
- OS Version | motionEyeOS 20190119

while the PC version user the following:

- motionEye Version | 0.35.1
- Motion Version | 3.2.12+git20140228

I have tried enabling traffic form camera on the firewall and didnt succeed. So i disabled the firewall and still didnt get any change. Anyone with any idea what is going on? 

I like the raspberry version but it has limitation on number of cameras and also the storage space. This is why I am trying to setup on PC.

ewanyande, 2019-05-08:
I am testing system using Milesight 3mp minidome camera.

karlmiles, 2019-05-24:
I had the exact same issue as yourself and resolved it by manually installing the Ubuntu Xenial package for motion from the project [github release page](https://github.com/Motion-Project/motion/releases).  Restart motioneye (I restarted motion as well to be sure), and it _should_ then work as expected.

JamBalaya56562, 2026-11-07:
Closing, as installing a newer Motion from the Motion releases page fixed the same problem for another user; the Motion package of Ubuntu 16.04 was old. Please open a new issue with your motionEye log if it still happens.

closed 2026-11-07 by JamBalaya56562