motioneye-project/motioneye #1270: After Upgrade: Setting current date using http: failed

kish1212, 2019-05-08:
Hello,
My MotionEyeOS was working perfectly till a few days ago, i upgraded to the latest version by downloading the latest version on my computer and flashing it with Etcher.
A few days it has been working correctly and now on 08.05.2019 at 11.15hrs i have been getting lots of issues with my MotionEyeOS. (1) Setting current date using http: failed, could not access the camera (2) After many restarts, it successfully captured the date, however it loads the MotionEyeOS page and keeps buffering and nothing is being displayed from the camera. (3) After several restart it got back to the camera but very slow and lagging and it freezes, even when refresh it fails to load the page. (4) Many times motion has been triggered but only few times email notification + photo is being sent. 
Note: I have been testing the internet at that specific time and it was normal, not even slow.
Can anyone please help me with that, I am attaching a few log files which may hopefully have detected the problem:

Thanks 

[boot.log](https://github.com/ccrisan/motioneye/files/3158682/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneye/files/3158683/dmesg.log)
[motion.log](https://github.com/ccrisan/motioneye/files/3158684/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/3158685/motioneye.log)
[wpa_supplicant.log](https://github.com/ccrisan/motioneye/files/3158686/wpa_supplicant.log)

jasaw, 2019-05-13:
MotionEye was unable to resolve host name.
>2019-05-08 11:44:54: [sendmail]    ERROR: failed to send mail: [Errno -3] Temporary failure in name resolution

>* Setting static DNS server to 8.8.8.8: done
# DNS server address is 192.168.8.1

I'm not sure what's happening here, why reported DNS server is different from the set DNS address.

kish1212, 2019-05-13:
> MotionEye was unable to resolve host name.
>
>
>
> I'm not sure what's happening here, why reported DNS server is different from the set DNS address.

Hello jasaw,

What do you think may be the problem? Me too i never experienced this issue before.

kish1212, 2019-05-13:
Hello everyone,

Have anyone experience this issue?
Is there any clue why this issue has arised?
Any help would be much appreciated, please.
Thanks

ccrisan, 2019-05-13:
That could be due to using two network adapters at the same time. The first one picks up the static configuration while the second one would use DHCP.

kish1212, 2019-05-14:
Hello ccrisan,
Thanks for your precious feedback.
How can that be possible? Any idea?
How can the problem be overcome?
Thanks

JamBalaya56562, 2026-11-08:
Closing, as this was on motionEyeOS: setting the date over HTTP is part of motionEyeOS, and the cause pointed out above was two network adapters in use at once. motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos), so please ask there if it still happens.

closed 2026-11-08 by JamBalaya56562