motioneye-project/motioneye #1272: Camera URL read-only?

PieBru, 2019-05-12:
Hi,
while learning motioneye and testing/configuring cameras I often need to replace the camera URL, but the URL field is defined once and is fixed read-only.
The only way seems to delete and reinsert the camera, but this way I loose all the other parameters, which in my case is a pain.
Thank you,
Piero

zagrim, 2019-05-13:
I agree that it would be nice for testing if one could change the URL also in the GUI. But maybe there's a good reason why it has been implemented this way?

Anyway, @PieBru, you could also manually edit the corresponding camera config file in `/etc/motioneye` directory (e.g. `/etc/motioneye/thread-1.conf` or `/etc/motioneye/camera-1.conf`), change the camera URL and then restart motioneye service.

zagrim, 2019-05-13:
Some ideas: 

- Add a feature for copying existing camera config and applying that on a newly added camera. That would allow easily hauling over all previously configured things to the new camera. (I noticed when trying out Shinobi before finding motionEye that it has this feature, and given the daunting amount of config parameters Shinobi has per camera that is a necessity there. In motionEye it's still manageable to copy the config manually, IMO, although being a bit tedious and error-prone)
- Allow changing the camera URL and it's parameters in the UI but not the host part. It would be easy to implement, I think, if the camera URL was just split into two fields in the UI. Perhaps make that only available to the admin user, if that somehow fits the design better?

PieBru, 2019-05-13:
Yes, I realized there are good reasons to keep the camera URL read-only to average/casual users.
A "Clone settings dropdown in the new camera form would be a useful feature.
Idea: It may be useful also to have some hints in the form or mouseover text, on how to manually tweak a camera settings. Online immediate hints are a precious resource for users not yet acquainted with the app logic.

PieBru, 2019-05-16:
Thanks @zagrim, I changed the cameras stream path directly in /etc/motion/camera-xx.conf and it works.
As I have a bounch of D-Link DCS-932 and 933, this way I experimented motioneye massively and quickly with the (unofficial) cameras image/stream URLs:
/image.jpg
/mjpeg.cgi
/video.cgi
etc.

Thanks, I leave up to you to close this issue.
Piero

ilovelinux, 2019-06-20:
Same of https://github.com/ccrisan/motioneye/issues/245

closed 2019-06-21 by ccrisan