motioneye-project/motioneye #1275: Saved movies no longer being auto-deleted as per settings

NginUS, 2019-05-13:
I'm on Linux Mint 18.3 (Ubuntu 16.04 LTS)
motionEye Version | 0.40
Motion Version | 4.2.2

Several days ago I lost all my configuration & had to start over, detailed in issue 1261 "Timestamp Issues".

What's happening now since then is that although I have the 'Movies' section of the MotionEye settings set to a 'Custom' retention threshold of 14 days [1], the WebUI's Movies Browser displays [2] 15-day old videos. The only sense I can make of this is that maybe it doesn't count the current day when it decides what to delete?

Furthermore, when I browse that directory in my File Manager, it shows [3] there being a few even older files that were recorded far beyond the 15-day limit- oddly for just 2 days with a significant gap. Note that these aren't made available through the WebUI's Movie Browser. They're owned by root same as the ones that are displayed, so I don't know how to make sense of that.

[1]
https://files.catbox.moe/ifiktp.png
[image]

[2]
https://files.catbox.moe/npbxpp.png
[image]

[3]
https://files.catbox.moe/worlg9.png
[image]

ccrisan, 2019-05-14:
Could you please attach your motioneye log here so we can see what the cleanup procedure has to say?

NginUS, 2019-05-15:
[motioneye.log](https://github.com/ccrisan/motioneye/files/3182389/motioneye.log)

FYI: I've since manually deleted the problem files/directories. Here's a log anyway- doesn't look so helpful though. :-(

NginUS, 2019-05-22:
I've changed the retention to 1 week, & waited to see what it would do.

What's happening now is that its deleting week-old directories like it should, except only the ones its created since I reconfigured everything. I wound up with a gap- some old ones remained that were well over a week old, but earlier week-old ones were deleted. Now I've manually deleted those very old ones which it wouldn't delete & all is well.

closed 2019-05-22 by NginUS