motioneye-project/motioneye #1277: When motion detection mask is enabled but empty, UI silently fails

CRImier, 2019-05-15:
```
May 14 22:34:08 debian meyectl[516]:    ERROR: integer division or modulo by zero
May 14 22:34:08 debian meyectl[516]: Traceback (most recent call last):
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
May 14 22:34:08 debian meyectl[516]:     raise_exc_info((type, value, traceback))
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1756, in wrapper
May 14 22:34:08 debian meyectl[516]:     result = method(self, *args, **kwargs)
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 290, in post
May 14 22:34:08 debian meyectl[516]:     self.set_config(camera_id)
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 205, in wrapper
May 14 22:34:08 debian meyectl[516]:     return func(self, *args, **kwargs)
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 522, in set_config
May 14 22:34:08 debian meyectl[516]:     set_camera_config(int(key), cfg, check_finished)
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 369, in set_camera_config
May 14 22:34:08 debian meyectl[516]:     local_config = config.motion_camera_ui_to_dict(ui_config, local_config)
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 952, in motion_camera_ui_to_dict
May 14 22:34:08 debian meyectl[516]:     capture_width, capture_height)
May 14 22:34:08 debian meyectl[516]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/utils.py", line 843, in build_editable_mask_file
May 14 22:34:08 debian meyectl[516]:     ny = mask_height = height * MASK_WIDTH / width  # number of rectangles
May 14 22:34:08 debian meyectl[516]: ZeroDivisionError: integer division or modulo by zero
```

zagrim, 2019-06-12:
What do you actually mean by "empty" mask file? How can a user end up in a situation where they face this issue?

hspaay, 2020-02-02:
Just FYI, I ran into this same issue. In my case I changed the camera ID in the config file but forgot to update the mask filename, so mask was enabled but no mask file was found.
The reason to change the camera ID in the config file is because it is not editable in the UI and I'd like them to be the same as my own camera numbering. 
Not a biggie, I'd assume that if you mess around with the config file by hand you're on your own.
It would be good to make the software more robust by checking if the mask file exists before calculating the width.
Thanks for this great software!

closed 2026-11-08 by Marijn0