motioneye-project/motioneye #1278: Wiki installation instruction

henkiejan1, 2019-05-15:
Hi there,

First of all: great project motion eye! Thank you. :) Also really simple to install. But i think there is a little fault in the instruction for installation on Unbuntu (for me Armbian Ubuntu for the Orange Pi One). Pillow won't install because i needed to install a few more libraries.

First i needed to install in PIP setuptools. So ik needed to do:
`sudo pip install setuptools`

There after pillow won't install. I found the solution to install the following libraries:

`sudo apt-get install libtiff5-dev libjpeg8-dev zlib1g-dev \
    libfreetype6-dev liblcms2-dev libwebp-dev libharfbuzz-dev libfribidi-dev \
    tcl8.6-dev tk8.6-dev python-tk`

This is also suggested in the instruction of [Pillow](https://pillow.readthedocs.io/en/stable/installation.html#basic-installation). I just hope it will make installation for other users much simpler. :)

CRImier, 2019-05-16:
Also, you can do `apt install python-pillow`, I highly suggest you do that if you don't want to compile it from scratch. Works for me with 0.4.0 on x86.

henkiejan1, 2019-05-16:
I go test `apt install python-pillow` quickly and you all hear of me :)

rigg77, 2019-05-26:
I had significant, repeatable issues with `pip install pillow` and `apt install python-pillow` resolved, and allowed me to complete the MotionEye setup. Suggest making `python-pillow` part of the dependencies instead of fetched with pip.

CRImier, 2019-05-26:
That's why I do it - compiling `pillow` from scratch can be quite a pain, what's with installing all the needed libraries and freeing enough RAM/disk for compiling. However, that's what `pip` seems to do most often (at least, on RPi, it's most likely there's no ARM wheel available by default). `apt install python-pillow` gives you a pre-compiled version, which is both very fast and very easy to install.

henkiejan1, 2019-06-09:
Your suggestion is great @CRImier ! Don forget to install `pip install motioneye --no-cache` otherwise i got the same error. However: works!

ccrisan, 2019-06-09:
I've updated the installation wiki pages with this remark. Thanks.

closed 2019-06-09 by ccrisan

alexxtasi, 2019-06-25:
could be helpful if in the wiki's last step (9), a note is added like:

---
**note:** If upgrade procedure fails, you can try using the `--no-cache` option:
- ```pip install motioneye --upgrade --no-cache```
---
at least it worked for me

CRImier, 2019-10-09:
also, you can install `pycurl` from `apt`: `apt install python-pycurl`. In the end, you might not actually need to build anything ;-P