motioneye-project/motioneye #1279: sv3c 1080p Cam setup as a webcam

JohnPaden, 2019-05-17:
motionEye Version | 0.40 - Motion Version | 3.2.12+git20140228 - OS Version | Ubuntu 16.04
I did manage to add a SV3C 1080p dome webcam via Add Camera / Network Camera / url: http://x.x.x.x/cgi-bin/ipcam.cgi/snap.cgi?&-getstream&-chn=2 once. Due to a change of the ip adress of the cam I had to set it up again. Now I only get the 'no signal' signal. 
No Idea whats wrong - the URL in a firefox browser shows a video stream. 
I am stuck - is this due to some update I did?

JohnPaden, 2019-05-19:
I reinstalled motioneye on Ubuntu 18 LTS. Now it works again. 
Issue closed.

closed 2019-05-19 by JohnPaden

SargFowler, 2021-02-02:
Thanks for the IP address hint! Saved me a bit of googling!