motioneye-project/motioneye #1281: Question on frequent connection lost

PieBru, 2019-05-18:
Hi, I've set up 16 cameras, and most of them appear "greyed" or "unavailable" every morning.

The cameras are accessed by hostname, which is always resolved ok by my OpenWrt main router. I suspect my problem is related to IP change due to DHCP leases renewal, which over time isn't guaranteed to give the same address to the same mac/hostname.
Also, not in my case as I use WDS APs and extenders, but using most commercials AP/extenders the MAC address of each device can change dynamically when the client (re)connects.
The problem seems amplified by shortening the DHCP lease period, but it remains random, thus not easy to debug.

A solution could be to assign fixed DHCP leases to all LAN cameras/devices, but in my case it would be a maintenance nightmare.

Do you have any hint?
MotionEye "reconnects" a camera if its IP is changed?

Thanks,
Piero

PieBru, 2019-05-18:
Hint: disabling and re-enabling the camera temporarily regains access:
Video device = OFF --> Apply --> Video device = ON, Apply.

The problem happens either with jpeg and mjpeg cameras.

Thanks
Piero

PieBru, 2019-06-02:
Update: I mitigated the problem by migrating my wifi network to 802.11s only, using OpenWrt everywhere.
This guarantees all MAC addresses remains "original", and makes the DHCP server happy.
Also, I adopted a 172.16/16 LAN, so every device shoud not change IP frequently.
This way the MotionEye "grey morning" problem is highly mitigated, but I suppose it isn't 100% fail-proof until the name-ip resolution is refreshed at least every few minutes.
Note: I hate extensively using fixed DHCP leases, which in my case carry a lot of work over time.
Or I missed something on the MotionEye side?
Thanks,
Piero

JamBalaya56562, 2026-11-07:
Closing, as you mitigated this on the network side, with MAC addresses and leases that stay the same, and found that turning the camera off and on again brings it back. If cameras still drop when their address changes with a current release, please open a new issue.

closed 2026-11-07 by JamBalaya56562