motioneye-project/motioneye #1282: Unable to add remote motioneyeos cameras

rohtua, 2019-05-18:
Hi, I've got 2 motioneyeos servers setup on a couple of Raspberry Pi's, one with the camera attached, the other acting as a host server to view the feed. The intention being to add more cameras over time. I've had the system working for a few months now. I've just updated motioneyeos on the host pi to the latest version and now my remote camera is gone. When I try to re-add the camera I can't I fill out all the details. I can view the camera through the motioneyeos interface for that specific Pi (and that works while I only have the one camera but defeats the purpose as and when I add more). 

When I fill out the details for the server the add button won't work and the box next to camera remains blank. I've tried updating to the dev release to see if there were any fixes there but the problem persists. The only way I've been able to re-add remote motioneyeos cameras is by using an older release of motioneyeos.

zagrim, 2019-06-12:
You are not telling the versions you upgraded from and to, but just a guess: Have you followed the instructions mentioned in [the release notes](https://github.com/ccrisan/motioneye/releases/tag/0.40)? You need to run configuration migration when upgrading from 0.39 to 0.4x.

JamBalaya56562, 2026-11-08:
Closing, as zagrim's question above went unanswered. Upgrading from 0.39 to 0.40 needs the configuration migration described in the [0.40 release notes](https://github.com/motioneye-project/motioneye/releases/tag/0.40), as in #1393, which matches a camera that disappeared after the upgrade. Both devices ran motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-08 by JamBalaya56562