motioneye-project/motioneye #1283: Saved movies in both .avi & .mp4 while set to HEVC (.mp4)

NginUS, 2019-05-19:
I recently changed the settings in Motioneye to save as HEVC (.mp4) from having it set previously as MPEG4 (.avi).

When I checked tonight I see it's recording in both formats, & it doesn't appear to be completely consistent. Sometimes its saved as both for the same timestamp & other times not.

Screenshot:

[image]

NginUS, 2019-05-22:
Is there a config file somewhere that might be wrongly causing duplicates to be saved like this? This is a significant issue for me- I don't have the kind of disk space to support this behavior. Would really appreciate a solution for this.

ilovelinux, 2019-06-19:
Hey @NginUS, still working on this problem? Can you send us your camera's config file? You can find it on `/etc/motioneye/name_of_the_camera.conf`

NginUS, 2019-06-19:
No, the camera was one of those cheap ones from China, & died soon after this thread was started. The app's been uninstalled since so I'm afraid I can't help here anymore.

JamBalaya56562, 2026-11-10:
As you wrote, the camera and the install are gone, so this cannot be checked any more. Closing; if anyone sees movies saved in two formats on current motionEye, please open a new issue with the camera config.

closed 2026-11-10 by JamBalaya56562